robotology-legacy/codyco-superbuild #46: Merging new_wbi_ID in master

traversaro, 2015-01-19:
Hi everyone @robotology/codyco-admins @robotology/codyco-developers ,
    finally we are ready to merge the version 0.2 of the codyco software (the one contained in the `new_wbi_ID` branches) in master. 

We will proceed shortly and the users shouldn’t notice particular differences, at least from the simulink (WBI-Toolbox) level. However there are a few changes that could cause problems, so please pay attention at the first time you update the superbuild.

To get more information about what have changed in v0.2, please check the README of WBI-Toolbox for the new version v0.2 : https://github.com/robotology-playground/WBI-Toolbox/blob/dd77704262242963fd67835e63ac04c2425f1f23/README.md 

If you where developing in C++ using the wholeBodyInterface, please check the migration guide for 
moving from the wbi v0.1 to wbi v0.2 : 
* https://github.com/robotology-playground/wholebodyinterface/blob/4f341fef3af28d6644e05f950c96adb9983e60a3/Migration.md 
* https://github.com/robotology-playground/yarp-wholebodyinterface/blob/be7a30b56a4d3451f3db153d9dbf4ea9fd31932b/Migration.md 

If for any reason you will need the legacy version of the codyco software, please note that the codyco-superbuild will have a special branch named `v0.1` with everything necessary to build the legacy version of the codyco software.

closed 2015-01-19 by traversaro

traversaro, 2015-01-19:
The long standing migration has completed: to update to the new version from an existing master just perform the usual superbuild update:
```
git pull
cd build
make update-all
make
```

***Migrate from new_wbi_ID to master***
For migrating from an existing new_wbi_ID installation, some steps should be taken. The superbuild has still some problems with dealing with different branches in superprojects, so
for switching from new_wbi_ID to master is reccomended to use the script for switching the branch in the subprojects automatically:
```
git pull
git checkout master
./script/switch-all.sh master  
cd build
make update-all
make
```

***Continue to use v0.1***
If you want to continue use the old v0.1 software (old master branch) switch the superbuild repo and all relevant repos to the v0.1 branch, This can be easy thanks to the script provided by @jeljaik to easily switch all interested repos to a given branch:
```
git pull
git checkout v0.1
./script/switch-all.sh v0.1
cd build
make update-all
make
```

azadm, 2015-01-20:
Is it going to be
````
xcodebuild -configuration Release
```` 
instead of
````
make update-all
make
````
if I use Xcode?

traversaro, 2015-01-20:
I guess that @francesco-romano can help on this!

traversaro, 2015-01-20:
From the documentation of the YCM (http://robotology.github.io/ycm/gh-pages/master/manual/ycm-superbuild.7.html#global-targets) I guess  you should use the UPDATE_ALL target. 
From the xcodebuild documentation ( https://developer.apple.com/library/mac/documentation/Darwin/Reference/ManPages/man1/xcodebuild.1.html ) I guess then it would be sufficient to run
```
xcodebuild -configuration Release -target UPDATE_ALL
```

azadm, 2015-01-20:
that command didn't work

traversaro, 2015-01-20:
We will have to wait some OS X user then... (unfortunately I don't have a OS X test machine).

francesco-romano, 2015-01-20:
Documentation of the super build is here: http://robotology.github.io/ycm/gh-pages/master/manual/ycm-superbuild.7.html#os-x

So the command is 
```bash
xcodebuild -target ALL_UPDATE
```

garimaiitd, 2016-02-12:
Hi, i am having problem with MATLAB simulations. I am getting error like:
failed assertion in b:\matlab\toolbox\stateflow\src\stateflow\utils\object.cpp at line 2853:
	object[id]!=NULL

I looked into the post dated 29th Jan 2015 Memory allocation error with holder block #69
but dont understand how to rectify this. 

Please help!

traversaro, 2016-02-12:
Hi @garimaiitd , I will open a new issue discussing  your problems, this is an old Pull Request ( https://help.github.com/articles/using-pull-requests/ ) that was merged more then one year ago, and thus is not the best place where to discuss.

garimaiitd, 2016-02-12:
hi, ok. thanks. Please let me know if you need more clarifications on the
problem i m facing.

-Garima

PR #47 (2015-01-19): Merge new_wbi_ID branch in master
Fix #46