robotology-legacy/codyco-superbuild #49: Various maintenance tasks for all subprojects

traversaro, 2015-01-23:
There are several tasks that should be done for all subprojects of the superbuild, for which we need volunteers ( : ) ). This tasks are also a good way of getting in touch with all the codebase and all the build system) : @naveenoid  @nunoguedelha @rbonghi . 
* [ ] Migrate all print of yarp related library/modules (mainly in yarp-wholebodyinterface, codyco-modules and WBI-Toolbox) to the new yDebug/yInfo yarp macros (similarly to what @randaz81 did in https://github.com/robotology/icub-main/commit/c89def02c9094a0d008a1077e24521232e69c13b ). This would enable a much easier reading of output of complex setup using the new yarp-logger . 
Subprojects to update (feel free to add):
 * [ ] codyco-modules 
 * [ ] yarp-wholebodyinterface
 * [ ] codyco-modules
 * [ ] WBI-Toolbox
 * [ ] paramHelp 
 * [ ] codyco-commons

* [ ] Fix travis on all subprojects. Travis is a really useful continuous integration tool for validating that no commit break the build of a given project. The major problem for the subprojects is to deal with dependencies, but we can rely on the superbuild for that (see what I did in iDynTree for an example (still a work in progress) : https://github.com/robotology-playground/idyntree/pull/15 ) 
Subprojects to update (feel free to add):
 * [ ] codyco-modules 
 * [ ] wholebodyinterface
 * [ ] yarp-wholebodyinterface
 * [ ] codyco-modules
 * [ ] codyco-commons

naveenoid, 2015-01-30:
I can do the first task for codyco-modules and yarp-wholebodyinterface and maybe WBI-Toolbox.

traversaro, 2015-01-30:
@naveenoid coffee allocated!

nunoguedelha, 2015-02-02:
I'll be dealing with the "paramHelp" and "codyco-commons" modules

jeljaik, 2015-02-03:
Done for WBI-Toolbox as in: https://github.com/robotology-playground/WBI-Toolbox/commit/7384d1d3e3c8951b6ecc8b198bbb51f2108226cc

nunoguedelha, 2015-02-11:
Going for 2nd task (Travis fix) regarding wholebodyinterface and yarp-wholebodyinterface

traversaro, 2018-04-18:
`codyco-modules` will soon be deprecated, except for a few devices. Closing the issue here as it is not useful anymore.

closed 2018-04-18 by traversaro