robotology-legacy/codyco-superbuild #53: Coding conventions that mirror dynamics notation

traversaro, 2015-02-06:
@jeljaik @naveenoid 
You had the same idea! Here the issue to implement it. : )

naveenoid, 2015-02-06:
I guess the 4 main points to start off this discussion :
1. subscripts and superscripts - using latex like convention but starting
at the most superscripted character and using underscores for the rest :
eg. w_T_b
2. Greek symbols : using english names of symbols like omega.
3. Treating subscripted symbols like joint velocity as qj(double character)
or as q_j(using rule 1 above).
4. As always, matrixes in capitals, vectors non capitalised.

---
Naveen Kuppuswamy, PhD
Post-doctoral Fellow,
Cognitive Humanoids Lab,
Department of Robotics, Brain and Cognitive Sciences (RBCS),
Istituto Italiano di Tecnologia, Genova, Italy

traversaro, 2018-04-19:
I think this is a good idea, but it is each project documentation duty to define this.

closed 2018-04-19 by traversaro