robotology-legacy/codyco-superbuild #54: yarpWholeBodyInterface.ini not existing

robertocalandra, 2015-02-09:
I installed from zero the codyco-superbuild.
When trying to run the wholeBodyDynamicsTree it seems that it does not find the yarpWholeBodyInterface.ini 

A file search showed that this file does not exists anywhere within the codyco-superbuild folder/subfolders.

~~~
icub@tomlinson:~/Desktop$ wholeBodyDynamicsTree --headV2 --autoconnect
||| clearing context
||| adding context [wholeBodyDynamicsTree]
||| configuring
||| no policy found
||| default config file specified as wholeBodyDynamicsTree.ini
||| checking [/home/icub/Desktop/wholeBodyDynamicsTree.ini] (pwd)
||| checking [/home/icub/.config/yarp/robots/iCubDarmstadt01] (robot YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/robots/iCubDarmstadt01] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/iCubDarmstadt01] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/iCubDarmstadt01] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/iCubDarmstadt01] (robot YARP_CONFIG_DIRS)
||| checking [robots/iCubDarmstadt01] (robot YARP_DATA_DIRS)
||| checking [/home/icub/software/codyco-superbuild/build/install/share/codyco/robots/iCubDarmstadt01] (robot YARP_DATA_DIRS)
||| checking [config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/icub/software/codyco-superbuild/build/install/share/codyco/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/icub/.config/yarp/contexts/wholeBodyDynamicsTree] (context YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/contexts/wholeBodyDynamicsTree] (context YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/contexts/wholeBodyDynamicsTree] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/contexts/wholeBodyDynamicsTree] (context YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/contexts/wholeBodyDynamicsTree] (context YARP_CONFIG_DIRS)
||| checking [contexts/wholeBodyDynamicsTree] (context YARP_DATA_DIRS)
||| checking [/home/icub/software/codyco-superbuild/build/install/share/codyco/contexts/wholeBodyDynamicsTree] (context YARP_DATA_DIRS)
||| found /home/icub/software/codyco-superbuild/build/install/share/codyco/contexts/wholeBodyDynamicsTree
||| checking [/home/icub/software/codyco-superbuild/build/install/share/codyco/contexts/wholeBodyDynamicsTree/wholeBodyDynamicsTree.ini] (context)
||| found /home/icub/software/codyco-superbuild/build/install/share/codyco/contexts/wholeBodyDynamicsTree/wholeBodyDynamicsTree.ini
yarp: Port /wholeBodyDynamicsTree/rpc:i active at tcp://10.0.0.51:10710
||| finding file [wbi_conf_file]
||| checking [/home/icub/Desktop/yarpWholeBodyInterface.ini] (pwd)
||| checking [/home/icub/software/codyco-superbuild/build/install/share/codyco/contexts/wholeBodyDynamicsTree/yarpWholeBodyInterface.ini] (context)
||| checking [/home/icub/.config/yarp/yarpWholeBodyInterface.ini] (YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/yarpWholeBodyInterface.ini] (YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/yarpWholeBodyInterface.ini] (YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/yarpWholeBodyInterface.ini] (YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/yarpWholeBodyInterface.ini] (YARP_CONFIG_DIRS)
||| checking [yarpWholeBodyInterface.ini] (YARP_DATA_DIRS)
||| checking [/home/icub/software/codyco-superbuild/build/install/share/codyco/yarpWholeBodyInterface.ini] (YARP_DATA_DIRS)
||| did not find yarpWholeBodyInterface.ini
yarp: cannot read from
[ERR] loadIdListFromConfig error: requested list ROBOT_DYNAMIC_MODEL_JOINTS not found
[ERR] loadIdListFromConfig error: requested list ROBOT_DYNAMIC_MODEL_JOINTS not found
[ERR] wholeBodyDynamicsModule: impossible to load wbiId joint list with name ROBOT_DYNAMIC_MODEL_JOINTS
Module failed to open
icub@tomlinson:~/Desktop$
~~~

rbonghi, 2015-02-09:
Did you set the environment variable **YARP_ROBOT_NAME** ?

I think not, because:
```
checking [/home/icub/software/codyco-superbuild/build/install/share/codyco/yarpWholeBodyInterface.ini] (YARP_DATA_DIRS)
||| 
```
Retry the command, an example:
```
YARP_ROBOT_NAME="icubGazeboSim" wholeBodyDynamicsTree --headV2 --autoconnect
```

jeljaik, 2015-02-10:
Since you are using the real robot  the environmental variable
YARP_ROBOT_NAME should be iCubHeidelberg01. Either set it up in your
~/.bashrc as:
export YARP_ROBOT_NAME="iCubDarmstadt01"

Or launch wholeBodyDynamicsTree as Bonghi was suggesting, with the right
robot name of course.

jeljaik, 2015-02-10:
... Which I just realized has not been created yet. In less than a couple
of hours (after lunch) first person available in the group will commit the
configuration file for your robot (although it's just a copy of those
already existing, e.g. iCubGenova01)

jeljaik, 2015-02-10:
Fixed in: https://github.com/robotology-playground/yarp-wholebodyinterface/commit/4a1a07280820a4a910fcc8338ec50b49ed8ee0e6
Please update yarp-wholeBodyInterface. 
Remember this is a `yarp-wholeBodyInterface` issue. The right project to post this issues is: https://github.com/robotology-playground/yarp-wholebodyinterface/issues

robertocalandra, 2015-02-10:
@rbonghi 
YARP_ROBOT_NAME was already properly set, so it is not the problem.

@jeljaik 
No, the commit does not fix this issue. 
In /home/icub/software/codyco-superbuild/build/install/share/codyco/... there is still no iCubDarmstadt01 folder

jeljaik, 2015-02-10:
That's because after updating `yarp-wholeBodyInterface` you also need to compile and install. Specifically: 

```bash
cd $CODYCO_SUPERBUILD_DIR
make yarpWholeBodyInterface
```

robertocalandra, 2015-02-10:
It is my understanding that using
~~~
make update-all
~~~
for the codyco-superbuild should both (update,) compile and install also the yarp-wholeBodyInterface, or am I wrong?

jeljaik, 2015-02-10:
No @robertocalandra, as explained in section https://github.com/robotology/codyco-superbuild#update, `make update-all` is the equivalent to `git pull` on each repository of the bundle. You should then do:

```bash
cd $CODYCO_SUPERBUILD_DIR
make yarpWholeBodyInterface
```

And then manually check the directory `$CODYCO_SUPERBUILD_DIR/install/share/codyco/robots/iCubDarmstadt01`  to double-check the file `yarpWholeBodyInterface.ini` has been correctly installed.

traversaro, 2015-02-12:
This should be fixed now, please reopen if the issue still persist.

closed 2015-02-12 by traversaro