robotology-legacy/codyco-superbuild #55: iCub model generation

traversaro, 2015-02-11:
(Actually a meta issue, but opening here because it affects mostly people in our team). 

For generating URDF models of the iCub, we rely now on two different systems:
* The old generation system based on iDyn data and paris meshes, available at https://github.com/robotology-playground/icub-model-generator
* The new generation system from actual CAD models converted to simmechanics file, described in http://wiki.icub.org/wiki/Creo_Mechanism_to_URDF . 

For now the only robot that can use the new generation system is iCubHeidelberg01, because for all the other robots we still don't have full simmechanisms model.

hu-yue, 2015-03-12:
Note to self: add symmetry tests to generated models.

traversaro, 2015-04-08:
Hi guys working on simulation @jeljaik @nunoguedelha @prashanthr05 , I tried to implement the changes (box collision, contact friction values, joint friction values that you did directly in the `icub-gazebo` repository in the `icub-model-generator` (i.e. the old generation system), in https://github.com/robotology-playground/icub-model-generator/commit/8fda11930c145d079a3720212cf68c894a779555 . 

The resulting generated model is the one in https://github.com/robotology-playground/icub-model-generator/blob/master/generated/gazebo_models/icubGazeboSim/icub.sdf . Can you verify that its matches the one in `icub-gazebo` and eventually merge it in the master branch of `icub-gazebo`? 
In this way the resulting equivalent urdf file (https://github.com/robotology-playground/icub-model-generator/blob/master/generated/gazebo_models/icubGazeboSim/icub_simulation.urdf) can be committed in the `yarpWholeBodyInterface` repo, so we can have simulated and control model that match.

traversaro, 2015-04-09:
I further debugged the model and committed the updated version in https://github.com/robotology-playground/icub-gazebo/commit/512478610e79a4a40a48fd190a5157631098640a . This should fix the mismatch in FT sensor orientations between the iCub simulated in Gazebo and the real robot. Furthermore now the gazebo model is matching the icubGazeboSim urdf in the `yarp-wholebodyinterface`. Please report any problem you may find.

traversaro, 2015-04-27:
cc @prashanthr05

traversaro, 2015-06-15:
* [ ] generate model for `iCubGenova02`

traversaro, 2016-05-05:
We should add a check for being sure that (at configuration 0) the frame that we use `l_sole`, `r_sole`, `root_link`, have the right rotations.

traversaro, 2016-06-07:
Models generated using the new system should be modified to contain the Denavit Hartenberg frames used by `iCubGui` for visualization. The frames are documented in http://wiki.icub.org/wiki/ICub_Model_naming_conventions .  For models generated with the v1 system those frames were there because the models were generated from the DH-like iDyn models, but for models generated with the v2 system, we need to add the frames ourselfs.

traversaro, 2016-06-22:
We need to regenerate the models for `iCubGenova02`/ `iCubDarmstad01` / `iCubGenova05` (i.e. all the models using the v2 model generation system) because the CAD model generated from CAD has been updated to include the wrist dofs (and to fix the frames of the accelerometers, see https://github.com/robotology/icub-support/issues/246). 

We need then to understand the local modification that we did to these generated models : 
 * `iCubGenova02` : https://github.com/robotology/yarp-wholebodyinterface/commits/master/app/robots/iCubDarmstadt01/model.urdf 
 * `iCubDarmstadt01` : https://github.com/robotology/yarp-wholebodyinterface/commits/master/app/robots/iCubDarmstadt01/model.urdf 
* `iCubGenova05` : https://github.com/robotology/yarp-wholebodyinterface/commits/master/app/robots/iCubGenova05/model.urdf

And port them back to the automatic model generation system, so we fully automatize the regeneration of the models if the CAD model changes again. 

cc @DanielePucci @francesco-romano @nunoguedelha @gabrielenava @S-Dafarra

traversaro, 2016-06-22:
Manual modifications to the code, note that `iCubGenova05` was just copied from `iCubDarmstadt01` : 
* **Backbattery removal** , https://github.com/robotology/yarp-wholebodyinterface/commit/fb5d6cadacc97de8748efe7ee7dd8abbedc8c988 :   `iCubDarmstadt01` was just copied from `iCubGenova02` manually changing the mass and center of mass of the `chest` link. We can probably automatize that using the [`assignedMasses` option in `simmechanics-to-urdf`](https://github.com/robotology/simmechanics-to-urdf#inertia-parameters) .
* **Adding `base_link` fake link** , https://github.com/robotology/yarp-wholebodyinterface/commit/5c22580db144ac3c72c555c826de2a67ca715ee2 : for a really old bug in KDL-based URDF parser  ( https://github.com/ros/robot_model/issues/6 ) there is a problem in models for which the root link is an actual link (i.e. with mass, inertias, etc etc). For this reason we tipically add a fake `base_link` link with 0 masses and inertia to workaround this problem. Our old KDL-based iDynTree software (still used by the WBI) needs this, so we need to add an option to automatically add it int `simmechanics-to-urdf`.
* **Fix direction of rotation of `torso_roll` joint** , https://github.com/robotology/yarp-wholebodyinterface/commit/0f32a69c6f2c20624962f2561bf97d109036090d : I must confess I totally forgot this fix. I remember some discussions about how the joint axis direction is processed in our pipeline, and I remember that the problem was relatively unsolvable (the direction is not directly accessible from CREO ), so we just need to make sure that the `reverseRotationAxes` parameters  in https://github.com/robotology/icub-gazebo/blob/master/iCubGenova02/iCubGenova02_simmechanics_options.yaml#L335 are correct. Ideally this should be checked by post-generation tests on the models, as suggested in https://github.com/robotology/codyco-superbuild/issues/55#issuecomment-217134635 . 
* **Add *_dh_frames**  https://github.com/robotology/yarp-wholebodyinterface/commit/b99d45c92e9e88510572d49a72f0646d6679ab20 : For internal links this will be addressed by https://github.com/robotology/codyco-superbuild/issues/55#issuecomment-224243988 , for end effectors we need to manually add them, I hope we can just add them as `XMLBlobs`. 
* **Add urdf-style force torque sensors**, https://github.com/robotology/yarp-wholebodyinterface/commit/94b1d20f5db6b42a5f1a48357f40c92deee1714b : this is needed by new non-KDL iDynTree sensors classes, this can be added either as an `XMLBlobs` or by augmenting the output for sensors.

traversaro, 2016-06-22:
Actually, the simmechanics XML still needs to be generated from the CREO files, and for that  I need the machine with Creo at IIT. 
For now I just started refactoring `icub-model-generator` : https://github.com/robotology-playground/icub-model-generator/tree/integrateNewWorkflow to support the new workflow.

traversaro, 2016-06-27:
I was successful in cleaning up the generation process in https://github.com/robotology-playground/icub-model-generator/tree/integrateNewWorkflow . Furthermore test have been added to avoid regression in thing such as the presence of sensors, the direction of axis and the presence of some important frames.

traversaro, 2016-06-29:
COM with backpack : 
~~~
<origin xyz="0.000188619190735 0.073091 -0.047238" rpy="0 0 0"/>
~~~
COM without backpack : 
~~~
 <origin xyz="0.000188619190735 0.023091 -0.013238" rpy="0 0 0"/>
~~~

traversaro, 2016-06-29:
Fixed most of the problems discussed in this issue in https://github.com/robotology-playground/icub-model-generator/pull/10/commits/b0b4686902be6f1c26914933b82409eaf0d505ed (except for the missing dh_frames ). 
Some steps are extremly hacky, but I am happy that are fully automated even if in a hacky way. 
The first automatically generated models are available in https://github.com/robotology-playground/icub-model-generator/pull/10/commits/b0b4686902be6f1c26914933b82409eaf0d505ed @DanielePucci .

traversaro, 2016-06-30:
Most of the problems related to iCub model generation have been fixed in https://github.com/robotology-playground/icub-model-generator/pull/10 . A lot still needs to be done, but it can be discussed directly in the icub-model-generator repo.

closed 2016-06-30 by traversaro

DanielePucci, 2016-07-04:
Thanks @traversaro, we will try to test the model as soon as possible.