robotology-legacy/codyco-superbuild #62: qpOASES integrated in the superbuild

traversaro, 2015-04-02:
Issue to solve: 
  * the matlab/simulink bindings build system is crazy and not based on cmake. 
   - using the new FindMatlab.cmake that we backported from CMake 3.3 to CMake 2.8.11 should simplify a lot the mex calibrtion
  * qpOASES does not export any library information 
   - YCM has already full support for building or finding qpOASES .

traversaro, 2015-04-02:
Added in this commit: https://github.com/robotology/codyco-superbuild/commit/b206a9aa2f64edf34766d7366fa81c8f9711444c .

traversaro, 2015-04-02:
cc @DanielePucci @francesco-romano

traversaro, 2015-04-03:
See also https://projects.coin-or.org/qpOASES/ticket/27 for a problem that we could want to solve in matlab bindings in OS X.

traversaro, 2015-04-03:
* [ ] if we add qpOASES as a superbuild dependency, we should add svn as a dependency, and mention it in the compilation.

naveenoid, 2015-05-11:
Having a problem with make update-all --> 

CMake Error at /home/naveenoid/codyco-superbuild/build/external/qpOASES/CMakeFiles/YCMTmp/qpOASES-gitclone.cmake:112 (message):
  Failed to clone repository:
  'https://github.com/robotology-playground/qpOASES.git'

make[3]: *** [external/qpOASES/CMakeFiles/YCMStamp/qpOASES-download] Error 1
make[2]: *** [CMakeFiles/qpOASES-update.dir/all] Error 2
make[1]: *** [CMakeFiles/update-all.dir/rule] Error 2
make: *** [update-all] Error 2

traversaro, 2015-05-11:
I hope it was a github glitch... can you try again after 5 minutes?

naveenoid, 2015-05-11:
nope..no change

traversaro, 2015-05-11:
Mh... as a temporary workaround you can try to disable qpOASES Compilation by disabling the `CODYCO_BUILD_QPOASES` CMake option.

iron76, 2015-06-24:
Yes, I recommend to set default value `CODYCO_BUILD_QPOASES` as `OFF` since I just experienced a compile error in macOS once enabled the `CODYCO_USES_MATLAB` option. Error was:
>  /Applications/MATLAB_R2012b.app/extern/include/tmwtypes.h:819:9: error: unknown type name 'char16_t'
 >  typedef char16_t CHAR16_T;

Solved by setting `OFF`the option `CODYCO_BUILD_QPOASES`.

closed 2018-03-08 by traversaro