robotology-legacy/codyco-superbuild #69: Tutorial nice to have on the codyco-superbuild

traversaro, 2015-05-08:
It would be nice to have an easy entry point (i.e. github README or doxygen home) for codyco users for the following use cases (they can be short tutorials or link to existing tutorials):
 * Use the simulink balancing controller using the Gazebo simulator,
 * Use the simulink balancing controller on the real robot,
 * Use the C++ torqueBalancing controller using the Gazebo simulator,
 * USe the C++ torqueBalancing controller using the real robot . 

For the torqueBalancing the existing readme probably does most of the job, we just need to link it. 

cc @francesco-romano @DanielePucci

francesco-romano, 2015-05-08:
C++ documentation is available here: https://github.com/robotology/codyco-modules/blob/master/src/modules/torqueBalancing/README.md

azadm, 2015-05-08:
Thanks for the documentations.
I tried to run the controller but when I run
````
wholeBodyDynamicsTree --autoconnect --assume_fixed l_foot_dh_frame
````
I get this error:

````
||| clearing context
||| adding context [wholeBodyDynamicsTree]
||| configuring
||| no policy found
||| default config file specified as wholeBodyDynamicsTree.ini
||| checking [/Users/azadm/wholeBodyDynamicsTree.ini] (pwd)
||| checking [/Users/azadm/Library/Application Support/yarp/config/robots/icubGazeboSim] (robot YARP_CONFIG_HOME)
||| checking [/Users/azadm/Library/Application Support/yarp/robots/icubGazeboSim] (robot YARP_DATA_HOME)
||| checking [/Library/Preferences/yarp/robots/icubGazeboSim] (robot YARP_CONFIG_DIRS)
||| checking [/etc/yarp/robots/icubGazeboSim] (robot YARP_CONFIG_DIRS)
||| checking [/Users/azadm/Documents/yarp/build/share/yarp/robots/icubGazeboSim] (robot YARP_DATA_DIRS)
||| checking [/Users/azadm/Documents/icub-main/build/share/iCub/robots/icubGazeboSim] (robot YARP_DATA_DIRS)
||| checking [/Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim] (robot YARP_DATA_DIRS)
||| found /Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim
||| checking [/Users/azadm/Documents/yarp/build/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/Users/azadm/Documents/icub-main/build/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/wholeBodyDynamicsTree.ini] (robot)
||| checking [/Users/azadm/Library/Application Support/yarp/config/contexts/wholeBodyDynamicsTree] (context YARP_CONFIG_HOME)
||| checking [/Users/azadm/Library/Application Support/yarp/contexts/wholeBodyDynamicsTree] (context YARP_DATA_HOME)
||| checking [/Library/Preferences/yarp/contexts/wholeBodyDynamicsTree] (context YARP_CONFIG_DIRS)
||| checking [/etc/yarp/contexts/wholeBodyDynamicsTree] (context YARP_CONFIG_DIRS)
||| checking [/Users/azadm/Documents/yarp/build/share/yarp/contexts/wholeBodyDynamicsTree] (context YARP_DATA_DIRS)
||| checking [/Users/azadm/Documents/icub-main/build/share/iCub/contexts/wholeBodyDynamicsTree] (context YARP_DATA_DIRS)
||| checking [/Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/contexts/wholeBodyDynamicsTree] (context YARP_DATA_DIRS)
||| found /Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/contexts/wholeBodyDynamicsTree
||| checking [/Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/contexts/wholeBodyDynamicsTree/wholeBodyDynamicsTree.ini] (context)
||| found /Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/contexts/wholeBodyDynamicsTree/wholeBodyDynamicsTree.ini
[INFO]assume_fixed option found, using  l_foot_dh_frame  as fixed link as a kinematic root instead of the imu. 
yarp: Port /wholeBodyDynamicsTree/rpc:i active at tcp://147.188.200.113:10039
||| finding file [wbi_conf_file]
||| checking [/Users/azadm/yarpWholeBodyInterface.ini] (pwd)
||| checking [/Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini] (robot)
||| found /Users/azadm/Documents/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini
RemoteControlBoard is ENABLING the writeStrict option for all commands
yarp: Port /wholeBodyDynamicsTree/torso/rpc:o active at tcp://147.188.200.113:10040
yarp: Port /wholeBodyDynamicsTree/torso/command:o active at tcp://147.188.200.113:10041
yarp: Port /wholeBodyDynamicsTree/torso/state:i active at tcp://147.188.200.113:10042
yarp: Port /wholeBodyDynamicsTree/torso/stateExt:i active at tcp://147.188.200.113:10043
Problem connecting to /icub/torso/rpc:i, is the remote device available?
Problem connecting to /icub/torso/command:i, is the remote device available?
Problem connecting to /icub/torso/state:o from /wholeBodyDynamicsTree/torso/state:i, is the remote device available?
*** Extended port /icub/torso/stateExt:o was not found on the controlBoardWrapper I'm connecting to. Falling back to compatibility beaviour
Updating to newer yarp and the usage of controlBoardWrapper2 is suggested***
yarpdev: ***ERROR*** driver <remote_controlboard> was found but could not open
[ERROR]Problems instantiating the device driver torso
[ERR] yarpWholeBodySensors::init() error: failing in opening encoders.
[ERROR]wholeBodyDynamicsTree : Error while initializing whole body estimator interface.Closing module 
Module failed to open
````

francesco-romano, 2015-05-08:
My fault.
WholeBodyDynamicsTree should be launched on the robot only. (torques are already provided by gazebo)

traversaro, 2015-05-08:
@francesco-romano leaving open until we have a clear and user friendly  path from the user entry point ( http://codyco.eu/software ? ) to the tutorials.

azadm, 2015-05-08:
Thanks @francesco-romano but it seems that the gains of the controller are for the real robot since the controller doesn't stabilize the robot in gazebo.

francesco-romano, 2015-05-08:
Is the `YARP_ROBOT_NAME` properly configured to point to the simulated robot?
What about the real time factor of the simulator?

Lastly: what do you mean by "stabilize" the robot? At the current state it can be possible to observe a little drift in simulation of some joints, but this should happen after some time.

azadm, 2015-05-08:
Yes, `YARP_ROBOT_NAME` is set correctly and it is not the problem of real time factor. Actually, the balancing controller in `WBIToolboxControllers` works fine on my machine but this one doesn't stabilize the robot meaning that it fails to keep the robot's balance.  It seems that the robot hits the joint limits somehow...

francesco-romano, 2015-05-08:
Here with gazebo 5.1 and the last version of `gazebo-yarp-plugins` and `icub-gazebo` everything seems fine (with fine defined as before: a part from a slow drift nothing exceptional happens)
Gains were taken from the simulink version.

azadm, 2015-05-08:
OK. Thanks @francesco-romano I'll try to update everything ...

traversaro, 2015-12-01:
cc @nunoguedelha I had some brainstorming here for how to provide entry points to our software, feel free to take part in the discussion.

serena-ivaldi, 2015-12-02:
I am super interested. I have a new student and I also formatted my machine. So we need to re-install everything. Could you quickly provide a list of the entry points for the codyco software? Is there documentation for all modules? As we dn't have iCub yet we can only test in simulation but it would be already something.

jeljaik, 2015-12-02:
Hi @serena-ivaldi,
In a nutshell... the current main entry point is the [README.md](https://github.com/robotology/codyco-superbuild) of `codyco-superbuild`.
The [libraries](https://github.com/robotology/codyco-superbuild#libraries) section lists all current available libraries and next to each one you find the link to their corresponding Project Page, where yet again, the documentation is found in their README file. 
The [main](https://github.com/robotology/codyco-superbuild#libraries) component of the project corresponds to modules and other software using the aforementioned libraries. In particular, [codyco-modules](https://github.com/robotology/codyco-modules) contains the main software modules of the project.
`codyco-superbuild` takes care of downloading all projects (each has its own repository), compiling them, and eventually updating them all if desired by the user. That's why, this is the starting point for any component of the project you might be interested in.

traversaro, 2016-01-13:
First link added in the doxygen home in https://github.com/robotology/codyco-superbuild/commit/5c9b72fa497854838b721727928aa8249abe72a8 .

traversaro, 2018-04-19:
Spiritual successor : https://github.com/robotology/robotology-superbuild/issues/65 .

closed 2018-04-19 by traversaro