robotology-legacy/codyco-superbuild #71: [CMake Error] file download hash mismatch

rlober, 2015-05-29:
Hey guys,

I just tried to recompile the HEAD of superbuild and got the following errors. I am not sure what they are referring to.

```
CMake Error at cmake/IncludeUrl.cmake:178 (file):
   file DOWNLOAD HASH mismatch

     for file: [/usr/local/src/robot/codyco-superbuild/build/CMakeFiles/ExternalProject.cmake]
       expected hash: [dcea39cc588f122668435b5be802119181f132be]
         actual hash: [2b3ad25b2fe758e8c5d45569824c4abe23b6c6bd]

 Call Stack (most recent call first):
   build/CMakeFiles/YCMEPHelper.cmake:108 (include_url)
   build/CMakeFiles/YCMEPHelper.cmake:186 (_ycm_include)
   build/CMakeFiles/YCMEPHelper.cmake:1415 (_ycm_setup)
   cmake/IncludeUrl.cmake:264 (include)
   cmake/YCMBootstrap.cmake:105 (include_url)
   CMakeLists.txt:53 (include)

 CMake Error at cmake/IncludeUrl.cmake:245 (message):
   include_url HASH mismatch

     for file: [/usr/local/src/robot/codyco-superbuild/build/CMakeFiles/ExternalProject.cmake]
       expected hash: [dcea39cc588f122668435b5be802119181f132be]
         actual hash: [2b3ad25b2fe758e8c5d45569824c4abe23b6c6bd]

 Call Stack (most recent call first):
   build/CMakeFiles/YCMEPHelper.cmake:108 (include_url)
   build/CMakeFiles/YCMEPHelper.cmake:186 (_ycm_include)
   build/CMakeFiles/YCMEPHelper.cmake:1415 (_ycm_setup)
   cmake/IncludeUrl.cmake:264 (include)
   cmake/YCMBootstrap.cmake:105 (include_url)
   CMakeLists.txt:53 (include)
```

traversaro, 2015-05-29:
cc @drdanz ?

traversaro, 2015-05-29:
In the meanwhile as a workaround you can install ycm directly from https://github.com/robotology/ycm , instead of bootstrapping it.

traversaro, 2015-05-29:
I don't understand what is happening because the IncludeUrl.cmake and YCMBootstrap.cmake did not changed, and we had our own copy of ExternalProject.cmake ...

rlober, 2015-05-29:
Hey @traversaro 

Where should I install YCM to?

drdanz, 2015-05-29:
Should be fixed, please try again

closed 2015-05-29 by drdanz

drdanz, 2015-05-29:
@traversaro Just for reference, what happened here is that I changed the ExternalProject.cmake file in YCM and did not update the sha1sum. YCM always checks for the sha1sum of the files to ensure that you are not downloading something broken.

rlober, 2015-06-01:
@drdanz 

Sorry for the delayed reply, but as you suspected the problem is fixed.

Thank you!