robotology-legacy/codyco-superbuild #72: [orocos-kdl] compile error

rlober, 2015-06-10:
Hey guys, I see that the build is failing but just in case it helps here is a compile error I am getting:

```
[ 21%] Skipping update step for 'orocos_kdl'
[ 21%] Performing configure step for 'orocos_kdl'
cmake version 2.8.11.2
Usage

  cmake [options] <path-to-source>
  cmake [options] <path-to-existing-build>

Options
  -C <initial-cache>          = Pre-load a script to populate the cache.
  -D <var>:<type>=<value>     = Create a cmake cache entry.
  -U <globbing_expr>          = Remove matching entries from CMake cache.
  -G <generator-name>         = Specify a makefile generator.
  -T <toolset-name>           = Specify toolset name if supported by
                                generator.
  -Wno-dev                    = Suppress developer warnings.
  -Wdev                       = Enable developer warnings.
  -E                          = CMake command mode.
  -i                          = Run in wizard mode.
  -L[A][H]                    = List non-advanced cached variables.
  --build <dir>               = Build a CMake-generated project binary tree.
  -N                          = View mode only.
  -P <file>                   = Process script mode.
  --find-package              = Run in pkg-config like mode.
  --graphviz=[file]           = Generate graphviz of dependencies.
  --system-information [file] = Dump information about this system.
  --debug-trycompile          = Do not delete the try_compile build tree.
                                Only useful on one try_compile at a time.
  --debug-output              = Put cmake in a debug mode.
  --trace                     = Put cmake in trace mode.
  --warn-uninitialized        = Warn about uninitialized values.
  --warn-unused-vars          = Warn about unused variables.
  --no-warn-unused-cli        = Don't warn about command line options.
  --check-system-vars         = Find problems with variable usage in system
                                files.
  --help-command cmd [file]   = Print help for a single command and exit.
  --help-command-list [file]  = List available listfile commands and exit.
  --help-commands [file]      = Print help for all commands and exit.
  --help-compatcommands [file]= Print help for compatibility commands.
  --help-module module [file] = Print help for a single module and exit.
  --help-module-list [file]   = List available modules and exit.
  --help-modules [file]       = Print help for all modules and exit.
  --help-custom-modules [file]= Print help for all custom modules and exit.
  --help-policy cmp [file]    = Print help for a single policy and exit.
  --help-policies [file]      = Print help for all policies and exit.
  --help-property prop [file] = Print help for a single property and exit.
  --help-property-list [file] = List available properties and exit.
  --help-properties [file]    = Print help for all properties and exit.
  --help-variable var [file]  = Print help for a single variable and exit.
  --help-variable-list [file] = List documented variables and exit.
  --help-variables [file]     = Print help for all variables and exit.
  --copyright [file]          = Print the CMake copyright and exit.
  --help,-help,-usage,-h,-H,/?= Print usage information and exit.
  --help-full [file]          = Print full help and exit.
  --help-html [file]          = Print full help in HTML format.
  --help-man [file]           = Print full help as a UNIX man page and exit.
  --version,-version,/V [file]= Show program name/version banner and exit.

Generators

The following generators are available on this platform:
  Unix Makefiles              = Generates standard UNIX makefiles.
  Ninja                       = Generates build.ninja files (experimental).
  CodeBlocks - Ninja          = Generates CodeBlocks project files.
  CodeBlocks - Unix Makefiles = Generates CodeBlocks project files.
  Eclipse CDT4 - Ninja        = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Unix Makefiles
                              = Generates Eclipse CDT 4.0 project files.
  KDevelop3                   = Generates KDevelop 3 project files.
  KDevelop3 - Unix Makefiles  = Generates KDevelop 3 project files.
  Sublime Text 2 - Ninja      = Generates Sublime Text 2 project files.
  Sublime Text 2 - Unix Makefiles
                              = Generates Sublime Text 2 project files.

[ 23%] Performing build step for 'orocos_kdl'
make[3]: *** No targets specified and no makefile found.  Stop.
make[2]: *** [external/orocos_kdl/CMakeFiles/YCMStamp/orocos_kdl-build] Error 2
make[1]: *** [CMakeFiles/orocos_kdl.dir/all] Error 2
make: *** [all] Error 2

```

traversaro, 2015-06-10:
Sorry, I was away and I did not noticed that the compilation was failing.

traversaro, 2015-06-10:
I reproduced the problem. This could be a problem also for iCub computers. cc @nunoguedelha

traversaro, 2015-06-10:
A workaround was commited in https://github.com/robotology/codyco-superbuild/commit/51fe03968f581b8dc6c862a6abbccd2166ade46e (use ycm v0.1.1 instead of master). I don't know what is causing this bug, probably is a regression in YCM.

mingxing-liu, 2015-06-10:
Hi, I pulled and now I'm using ycm v0.1.1, but I got the following error:
-- Found YCM: /home/codyco/icub/software/src/codyco-superbuild/build/install/share/YCM (found version "0.1.1")
CMake Error at CMakeLists.txt:77 (include):
  include could not find load file:
    AddInstallRPATHSupport
CMake Error at CMakeLists.txt:78 (add_install_rpath_support):
  Unknown CMake command "add_install_rpath_support".
-- Configuring incomplete, errors occurred!
make[2]: *** [libraries/wholeBodyInterface/CMakeFiles/YCMStamp/wholeBodyInterface-configure] Error 1
make[1]: *** [CMakeFiles/wholeBodyInterface.dir/all] Error 2
make: *** [all] Error 2

traversaro, 2015-06-10:
Sorry... it should be fixed with https://github.com/robotology/codyco-superbuild/commit/60594d11558fdfc0bb9e3992e1474004390659bf .

mingxing-liu, 2015-06-10:
It works now. Thank you.

rlober, 2015-06-10:
Thanks man!

closed 2015-06-10 by rlober