robotology-legacy/codyco-superbuild #74: problem in building codyco-superbuild

azadm, 2015-06-19:
Here is the error that I get when I build codyco-superbuild on OSX:

````
** BUILD FAILED **

The following build commands failed:
	Ld /Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/bin/Release/RedBallGraspDemo normal x86_64
(1 failure)
make: *** [/Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/CMakeFiles/YCMStamp/Release/codyco-modules-build] Error 65
Command /bin/sh failed with exit code 2

** BUILD FAILED **

The following build commands failed:
	PhaseScriptExecution CMake\ Rules build/codyco-superbuild.build/Release/codyco-modules.build/Script-D02AEA0CC4184C09AF0C31A9.sh
(1 failure)

````

traversaro, 2015-06-19:
I have re-started an os x build in Travis to see if the same problem can be reproduced there.

azadm, 2015-06-19:
Yes, I have. I removed the build directory and made it again but the problem is still there.

traversaro, 2015-06-23:
Have you tried to build the superbuild with `make` instead of `xcodebuild` ? 
Travis build are ok on OS X ( https://travis-ci.org/robotology/codyco-superbuild/builds/67891396 ) .

francesco-romano, 2015-07-08:
@azadm did you manage to build?
If not, please reopen the issue.

azadm, 2015-07-08:
I actually haven't tried with travis build but I tried to reinstall it again and now I get a new error:

````
Users/azadm/Documents/codyco-superbuild/main/codyco-modules/src/modules/wholeBodyDynamicsTree/src/wholeBodyDynamics_IDLServer.cpp:285:5: error: use of undeclared identifier 'yError'
    yError("Missing server method '%s'?","bool wholeBodyDynamics_IDLServer::calib(const std::string& calib_code, const int32_t nr_of_samples)");
    ^
/Users/azadm/Documents/codyco-superbuild/main/codyco-modules/src/modules/wholeBodyDynamicsTree/src/wholeBodyDynamics_IDLServer.cpp:295:5: error: use of undeclared identifier 'yError'
    yError("Missing server method '%s'?","bool wholeBodyDynamics_IDLServer::calibStanding(const std::string& calib_code, const int32_t nr_of_samples)");
    ^
/Users/azadm/Documents/codyco-superbuild/main/codyco-modules/src/modules/wholeBodyDynamicsTree/src/wholeBodyDynamics_IDLServer.cpp:305:5: error: use of undeclared identifier 'yError'
    yError("Missing server method '%s'?","bool wholeBodyDynamics_IDLServer::calibStandingLeftFoot(const std::string& calib_code, const int32_t nr_of_samples)");
    ^
/Users/azadm/Documents/codyco-superbuild/main/codyco-modules/src/modules/wholeBodyDynamicsTree/src/wholeBodyDynamics_IDLServer.cpp:315:5: error: use of undeclared identifier 'yError'
    yError("Missing server method '%s'?","bool wholeBodyDynamics_IDLServer::calibStandingRightFoot(const std::string& calib_code, const int32_t nr_of_samples)");
    ^
/Users/azadm/Documents/codyco-superbuild/main/codyco-modules/src/modules/wholeBodyDynamicsTree/src/wholeBodyDynamics_IDLServer.cpp:325:5: error: use of undeclared identifier 'yError'
    yError("Missing server method '%s'?","bool wholeBodyDynamics_IDLServer::resetOffset(const std::string& calib_code)");
    ^
/Users/azadm/Documents/codyco-superbuild/main/codyco-modules/src/modules/wholeBodyDynamicsTree/src/wholeBodyDynamics_IDLServer.cpp:335:5: error: use of undeclared identifier 'yError'
    yError("Missing server method '%s'?","bool wholeBodyDynamics_IDLServer::quit()");
    ^
/Users/azadm/Documents/codyco-superbuild/main/codyco-modules/src/modules/wholeBodyDynamicsTree/src/wholeBodyDynamics_IDLServer.cpp:345:5: error: use of undeclared identifier 'yError'
    yError("Missing server method '%s'?","bool wholeBodyDynamics_IDLServer::resetSimpleLeggedOdometry(const std::string& initial_world_frame, const std::string& initial_fixed_link)");
    ^
/Users/azadm/Documents/codyco-superbuild/main/codyco-modules/src/modules/wholeBodyDynamicsTree/src/wholeBodyDynamics_IDLServer.cpp:355:5: error: use of undeclared identifier 'yError'
    yError("Missing server method '%s'?","bool wholeBodyDynamics_IDLServer::changeFixedLinkSimpleLeggedOdometry(const std::string& new_fixed_link)");
    ^
8 errors generated.

** BUILD FAILED **

The following build commands failed:
	CompileC /Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/src/modules/wholeBodyDynamicsTree/codyco-modules.build/Release/wholeBodyDynamicsTree.build/Objects-normal/x86_64/wholeBodyDynamics_IDLServer.o src/modules/wholeBodyDynamicsTree/src/wholeBodyDynamics_IDLServer.cpp normal x86_64 c++ com.apple.compilers.llvm.clang.1_0.compiler
(1 failure)
make: *** [/Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/CMakeFiles/YCMStamp/Release/codyco-modules-build] Error 65

** BUILD FAILED **

The following build commands failed:
	PhaseScriptExecution CMake\ Rules build/codyco-superbuild.build/Release/codyco-modules.build/Script-26940021FD554FEAA4750B95.sh
(1 failure)

````

traversaro, 2015-07-08:
How did you installed yarp?

azadm, 2015-07-08:
I reinstalled yarp (with Xcode) and now I get this error:

````
Undefined symbols for architecture x86_64:
  "yarp::os::LockGuard::LockGuard(yarp::os::Mutex&)", referenced from:
      codyco::ManagerThread::threadInit() in ManagerThread.o
      codyco::ManagerThread::run() in ManagerThread.o
      codyco::ManagerThread::enableGrasping() in ManagerThread.o
      codyco::ManagerThread::disableGrasping() in ManagerThread.o
  "yarp::os::LockGuard::~LockGuard()", referenced from:
      codyco::ManagerThread::threadInit() in ManagerThread.o
      codyco::ManagerThread::run() in ManagerThread.o
      codyco::ManagerThread::enableGrasping() in ManagerThread.o
      codyco::ManagerThread::disableGrasping() in ManagerThread.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

** BUILD FAILED **

The following build commands failed:
	Ld /Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/bin/Release/torqueGraspManager normal x86_64
(1 failure)
make: *** [/Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/CMakeFiles/YCMStamp/Release/codyco-modules-build] Error 65
Showing first 200 notices only
Command /bin/sh failed with exit code 2

** BUILD FAILED **

The following build commands failed:
	PhaseScriptExecution CMake\ Rules build/codyco-superbuild.build/Release/codyco-modules.build/Script-5BED9999C0BC4DF5A8FE29DE.sh
(1 failure)

````

You think it is something wrong with my yarp installation?

francesco-romano, 2015-07-08:
Can you post the result of this command run in the yarp source directory? 

```
 git log | head 
```

azadm, 2015-07-08:
````
dynamic200-113:yarp azadm$ git log | head
commit 4ee83c2eafee4c67cc71506a6dfa14452b6a9060
Merge: b5d7a18 de19932
Author: Alberto Cardellino <alberto.cardellino@iit.it>
Date:   Wed Jul 8 13:53:01 2015 +0200

    Merge pull request #527 from robotology/setRefTorque_group
    
    Implementation of setRefTorque group

commit de19932e10bcd9a6c6dba4fb375d42ab3f6f2bd7

````

francesco-romano, 2015-07-08:
I'm a bit lost.
Can you try to post the compilation log of the `torqueGraspManager` (so the log with the options passed to the compiler. It should be just before the error).

azadm, 2015-07-08:
this one:

````
Ld /Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/bin/Release/torqueGraspManager normal x86_64
    cd /Users/azadm/Documents/codyco-superbuild/main/codyco-modules
    export MACOSX_DEPLOYMENT_TARGET=10.9
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -arch x86_64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.9.sdk -L/Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/bin/Release -F/Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/bin/Release -filelist /Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/src/modules/torqueGraspManager/codyco-modules.build/Release/torqueGraspManager.build/Objects-normal/x86_64/torqueGraspManager.LinkFileList -mmacosx-version-min=10.9 -Wl,-search_paths_first -Wl,-headerpad_max_install_names /Users/azadm/Documents/icub-main/build/lib/libctrlLib.a /Users/azadm/Documents/icub-main/build/lib/libicubmod.a /usr/local/lib/libgsl.dylib /usr/local/lib/libgslcblas.dylib /Users/azadm/Documents/yarp/build/lib/Release/libYARP_math.2.3.64.3.dylib /Users/azadm/Documents/yarp/build/lib/Release/libYARP_dev.2.3.64.3.dylib /Users/azadm/Documents/yarp/build/lib/Release/libYARP_name.2.3.64.3.dylib /Users/azadm/Documents/yarp/build/lib/Release/libYARP_init.2.3.64.3.dylib /Users/azadm/Documents/yarp/build/lib/Release/libYARP_sig.2.3.64.3.dylib /Users/azadm/Documents/yarp/build/lib/Release/libYARP_OS.2.3.64.3.dylib -Xlinker -dependency_info -Xlinker /Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/src/modules/torqueGraspManager/codyco-modules.build/Release/torqueGraspManager.build/Objects-normal/x86_64/torqueGraspManager_dependency_info.dat -o /Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/bin/Release/torqueGraspManager

````

francesco-romano, 2015-07-08:
Yes, but also, just before that there should be a `CompileC` part.

Also, are you compiling in a clean build or an old one?
Do you have the same issues also for debug compilation or only in release?

azadm, 2015-07-08:
````
CompileC /Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/src/modules/torqueGraspManager/codyco-modules.build/Release/torqueGraspManager.build/Objects-normal/x86_64/main.o src/modules/torqueGraspManager/src/main.cpp normal x86_64 c++ com.apple.compilers.llvm.clang.1_0.compiler
    cd /Users/azadm/Documents/codyco-superbuild/main/codyco-modules
    export LANG=en_US.US-ASCII
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -x c++ -arch x86_64 -fmessage-length=0 -fdiagnostics-show-note-include-stack -fmacro-backtrace-limit=0 -Wno-trigraphs -fpascal-strings -O3 -Wno-missing-field-initializers -Wno-missing-prototypes -Wno-return-type -Wno-non-virtual-dtor -Wno-overloaded-virtual -Wno-exit-time-destructors -Wno-missing-braces -Wparentheses -Wswitch -Wno-unused-function -Wno-unused-label -Wno-unused-parameter -Wno-unused-variable -Wunused-value -Wno-empty-body -Wno-uninitialized -Wno-unknown-pragmas -Wno-shadow -Wno-four-char-constants -Wno-conversion -Wno-constant-conversion -Wno-int-conversion -Wno-bool-conversion -Wno-enum-conversion -Wno-shorten-64-to-32 -Wno-newline-eof -Wno-c++11-extensions -DCMAKE_INTDIR=\"Release\" -DYARP_NO_DEPRECATED_WARNINGS -D_REENTRANT -DCODYCO_USES_URDFDOM -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.9.sdk -fasm-blocks -fstrict-aliasing -Wdeprecated-declarations -Winvalid-offsetof -mmacosx-version-min=10.9 -Wno-sign-conversion -I/Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/bin/Release/include -I/Users/azadm/Documents/codyco-superbuild/main/codyco-modules/src/modules/torqueGraspManager/include/codyco -I/Users/azadm/Documents/icub-main/src/libraries/ctrlLib/include -I/usr/local/include -I/Users/azadm/Documents/yarp/build/generated_include -I/Users/azadm/Documents/yarp/src/libYARP_OS/include -I/Users/azadm/Documents/yarp/src/libYARP_sig/include -I/Users/azadm/Documents/yarp/src/libYARP_math/include -I/Users/azadm/Documents/yarp/src/libYARP_dev/include -I/Users/azadm/Documents/yarp/src/libYARP_name/include -I/Users/azadm/Documents/yarp/src/libYARP_manager/include -I/Users/azadm/Documents/yarp/src/libYARP_logger/include -I/Users/azadm/Documents/yarp/src/yarpserver/include -I/Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/src/modules/torqueGraspManager/codyco-modules.build/Release/torqueGraspManager.build/DerivedSources/x86_64 -I/Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/src/modules/torqueGraspManager/codyco-modules.build/Release/torqueGraspManager.build/DerivedSources -Wmost -Wno-four-char-constants -Wno-unknown-pragmas -F/Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/bin/Release -DNDEBUG -MMD -MT dependencies -MF /Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/src/modules/torqueGraspManager/codyco-modules.build/Release/torqueGraspManager.build/Objects-normal/x86_64/main.d --serialize-diagnostics /Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/src/modules/torqueGraspManager/codyco-modules.build/Release/torqueGraspManager.build/Objects-normal/x86_64/main.dia -c /Users/azadm/Documents/codyco-superbuild/main/codyco-modules/src/modules/torqueGraspManager/src/main.cpp -o /Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/src/modules/torqueGraspManager/codyco-modules.build/Release/torqueGraspManager.build/Objects-normal/x86_64/main.o

Ld /Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/bin/Release/torqueGraspManager normal x86_64
    cd /Users/azadm/Documents/codyco-superbuild/main/codyco-modules
    export MACOSX_DEPLOYMENT_TARGET=10.9
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -arch x86_64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.9.sdk -L/Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/bin/Release -F/Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/bin/Release -filelist /Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/src/modules/torqueGraspManager/codyco-modules.build/Release/torqueGraspManager.build/Objects-normal/x86_64/torqueGraspManager.LinkFileList -mmacosx-version-min=10.9 -Wl,-search_paths_first -Wl,-headerpad_max_install_names /Users/azadm/Documents/icub-main/build/lib/libctrlLib.a /Users/azadm/Documents/icub-main/build/lib/libicubmod.a /usr/local/lib/libgsl.dylib /usr/local/lib/libgslcblas.dylib /Users/azadm/Documents/yarp/build/lib/Release/libYARP_math.2.3.64.3.dylib /Users/azadm/Documents/yarp/build/lib/Release/libYARP_dev.2.3.64.3.dylib /Users/azadm/Documents/yarp/build/lib/Release/libYARP_name.2.3.64.3.dylib /Users/azadm/Documents/yarp/build/lib/Release/libYARP_init.2.3.64.3.dylib /Users/azadm/Documents/yarp/build/lib/Release/libYARP_sig.2.3.64.3.dylib /Users/azadm/Documents/yarp/build/lib/Release/libYARP_OS.2.3.64.3.dylib -Xlinker -dependency_info -Xlinker /Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/src/modules/torqueGraspManager/codyco-modules.build/Release/torqueGraspManager.build/Objects-normal/x86_64/torqueGraspManager_dependency_info.dat -o /Users/azadm/Documents/codyco-superbuild/build/main/codyco-modules/bin/Release/torqueGraspManager
Undefined symbols for architecture x86_64:
  "yarp::os::LockGuard::LockGuard(yarp::os::Mutex&)", referenced from:
[... cut ...]
      codyco::ManagerThread::disableGrasping() in ManagerThread.o
  "yarp::os::LockGuard::~LockGuard()", referenced from:
      codyco::ManagerThread::threadInit() in ManagerThread.o
      codyco::ManagerThread::run() in ManagerThread.o
      codyco::ManagerThread::enableGrasping() in ManagerThread.o
      codyco::ManagerThread::disableGrasping() in ManagerThread.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

** BUILD FAILED **

````

francesco-romano, 2015-07-08:
We can try one last thing:
Open the Xcode project.
Go to file `ManagerThread.cpp` line 1205
Then right click on `LockGuard` and select in the menu `Jump to Definition`.
Can you please copy and paste here the opened file?

azadm, 2015-07-08:
it doesn't work...it says symbol not found!

francesco-romano, 2015-07-09:
Try the same procedure on the inclusion of the `LockGuard` (line 4, same file)

azadm, 2015-07-09:
It's the same. I tried with all of them in the file

azadm, 2015-07-09:
I reinstalled yarp with make command instead of using Xcode and now it's solved.

closed 2015-07-09 by azadm