robotology-legacy/codyco-superbuild #75: Building dylibs in superbuild.

rlober, 2015-06-23:
So I am in the process of breaking up `ISIR modules`/`ocra-wbi-plugins` into a set of shared libs and a set of modules. Currently I am just testing this new architecture on my local machine. In my root dir for each of the shared lib projects I have the lines:

```cmake
INSTALL(TARGETS ${PROJECTNAME} DESTINATION lib)
INSTALL(DIRECTORY include DESTINATION include/${PROJECTNAME})
```

**The issue:**
In `$CODYCO_SUPERBUILD_ROOT/build` when I run `make` the libs built and are copied over to `$CODYCO_SUPERBUILD_ROOT/build/install/lib` but the headers are not copied to `$CODYCO_SUPERBUILD_ROOT/build/install/include/[nameOfLib]`. 

Because I am **not** running `make install` it makes sense that the `INSTALL(...)` lines wouldn't be executed, however it is odd that one of them is executed `INSTALL(TARGETS ${PROJECTNAME} DESTINATION lib)`, but not the other. My guess is that this has to do with upstream CMake directives from superbuild, but I was hoping someone could clarify this for me.

traversaro, 2015-06-23:
Note that the `make` command of the superbuild is also executing the install target for each subproject, see http://robotology.github.io/ycm/gh-pages/master/manual/ycm-superbuild.7.html#ycm-superbuild-manual-for-basic-users . 
Hence, it should install the headers of the lib as well.. so this behavior is definitely strange. 
Have you checked if the headers are correctly installed if you do `make install` directly on the build directory of the project (that should be in `$CODYCO_SUPERBUILD_ROOT/build/$COMPONENT/$PROJECT` ).

rlober, 2015-06-23:
When I run `make install` directly on the project it does install the headers as expected.

traversaro, 2015-06-23:
During the superbuild run is printing something? For example, for codycoCommons the superbuild output is:
~~~
-- Install configuration: ""
-- Up-to-date: /home/pegua/src/codyco-superbuild/build/install/lib/libcodycocommons.so.0.0.1
-- Up-to-date: /home/pegua/src/codyco-superbuild/build/install/lib/libcodycocommons.so
-- Up-to-date: /home/pegua/src/codyco-superbuild/build/install/include/codyco/Commons.h
-- Up-to-date: /home/pegua/src/codyco-superbuild/build/install/include/codyco/Utils.h
-- Up-to-date: /home/pegua/src/codyco-superbuild/build/install/include/codyco/LockGuard.h
-- Up-to-date: /home/pegua/src/codyco-superbuild/build/install/include/codyco/MathUtils.h
-- Up-to-date: /home/pegua/src/codyco-superbuild/build/install/include/codyco/ControllerThread.h
-- Up-to-date: /home/pegua/src/codyco-superbuild/build/install/include/codyco/ModelParsing.h
-- Up-to-date: /home/pegua/src/codyco-superbuild/build/install/lib/cmake/codycoCommons/codycoCommonsConfigVersion.cmake
-- Up-to-date: /home/pegua/src/codyco-superbuild/build/install/lib/cmake/codycoCommons/codycoCommonsConfig.cmake
-- Up-to-date: /home/pegua/src/codyco-superbuild/build/install/lib/cmake/codycoCommons/codycoCommonsTargets.cmake
-- Installing: /home/pegua/src/codyco-superbuild/build/install/lib/cmake/codycoCommons/codycoCommonsTargets-noconfig.cmake
~~~
Where you can see that is installing headers, libraries and cmake configuration file. What is the output for your lib?

rlober, 2015-06-23:
Ok, so the issue was two fold:
1. The modules, which now use the libs have a linking error and were blocking the install step.
2. The `INSTALL` directive was not totally correct.

Using,
```cmake 
INSTALL(DIRECTORY ${PROJECT_SOURCE_DIR}/include/${PROJECTNAME} DESTINATION include)
```
and turning off the modules to be built, I get the following output:

```
[ 91%] Performing build step for 'ocra-wbi-plugins'
[ 50%] Built target ocraWbiPlugins
[100%] Built target taskSequences
[ 91%] Performing install step for 'ocra-wbi-plugins'
[ 50%] Built target ocraWbiPlugins
[100%] Built target taskSequences
Install the project...
-- Install configuration: ""
-- Up-to-date: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/lib/libocraWbiPlugins.so
-- Installing: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/ocraWbiPlugins
-- Up-to-date: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/ocraWbiPlugins/ocraWbiModel.h
-- Up-to-date: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/ocraWbiPlugins/ocraWbiUtil.h
-- Up-to-date: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/lib/libtaskSequences.so
-- Installing: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/taskSequences
-- Up-to-date: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/taskSequences/sequenceLibrary.h
-- Up-to-date: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/taskSequences/sequenceTools.h
-- Installing: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/taskSequences/sequences
-- Up-to-date: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/taskSequences/sequences/OrientationTest.h
-- Up-to-date: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/taskSequences/sequences/NominalPose.h
-- Up-to-date: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/taskSequences/sequences/LeftRightHandReach.h
-- Up-to-date: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/taskSequences/sequences/LeftHandReach.h
-- Up-to-date: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/taskSequences/sequences/TrajectoryTrackingTest.h
-- Up-to-date: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/taskSequences/sequences/FloatingBaseEstimationTests.h
-- Up-to-date: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/taskSequences/sequences/JointTest.h
-- Up-to-date: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/taskSequences/sequences/InitialPoseHold.h
-- Up-to-date: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/taskSequences/sequences/PoseTest.h
-- Up-to-date: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/taskSequences/sequences/FixedBaseMinimalTasks.h
-- Up-to-date: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/taskSequences/sequences/CartesianTest.h
-- Up-to-date: /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/include/taskSequences/sequences/FloatingBaseMinimalTasks.h
[ 91%] Completed 'ocra-wbi-plugins'
[ 94%] Built target ocra-wbi-plugins

```

Although I am still having compilation issues, I believe they are unrelated to superbuild so this issue is closed. Thanks for the help @traversaro !

closed 2015-06-23 by rlober

traversaro, 2015-06-23:
In the past we prepared some examples of CMake structure (using YCM) for a repo containing a library, I did not checked their state with respect to evolution of YCM, but if you are interested:
 * https://github.com/robotology-playground/lib-template-cmake
 * https://github.com/tacman-fp7/template-pkg