robotology-legacy/codyco-superbuild #76: RPC Calib on WholeBodyDynamicsTree fails

rueckert, 2015-07-06:
with 

[INFO]wholeBodyDynamicsTree : calibration for  arms requested 
[INFO] wholeBodyDynamicsThread::calibrateOffset called with code arms
wholeBodyDynamicsThread::decodeCalibCode error: requested feet calibration, but arms FT sensor are not available.
[ERR] wholeBodyDynamicsThread::decodeCalibCode failed

from wholeBodyDynamicsThread.cpp:604 where the arms_fts vector is never set and thus has a zero size. 

@traversaro 

Thank's for your help. Elmar and Alex

traversaro, 2015-07-06:
Which command are you sending to the `wholeBodyDynamicsTree` rpc?

traversaro, 2015-07-06:
If you are sending `calib arms`, then I see the bug, as a temporary workaround you can try a `calib all`.

rueckert, 2015-07-09:
@traversaro Calib all works

closed 2016-01-29 by francesco-romano