robotology-legacy/codyco-superbuild #77: Road to BERDY

traversaro, 2015-07-17:
I open this issue to list all the things we are missing to reach an implementation of the BERDY estimation framework running on the real robot. I am currently working for a paper on the matlab version of BERDY, contained in https://github.com/iron76/bnt_time_varying , so now I am getting some insight on the problem. I will try to outline my idea on the topic. 

I started writing but it began to be very long, so I move what I wanted to discuss to a wiki page: https://github.com/robotology/codyco-superbuild/wiki/Road-to-BERDY . 

Everyone should be able to edit it, for now I just drafted a preliminary objective and a brief list of milestones (intermediate goals). I will be happy to receive early feedback on that, and I have some questions related to the intermediate goal that we may want to discuss : 
 * Currently the floating base estimation scenarios based on BERDY are quite in the future, because they require the implementation of quite a lot of different techniques (dynamics MAP estimation, and only following that the EKF estimation of robot kinematic state). Unfortunately the floating base estimation is quite a critical issue on the control side, so perhaps it may be necessary to have another set of milestones related to floating base estimation that don't use BERDY techniques, but more classical ones, for example based on the @jeljaik existing code. This would be nice even for gradually increasing complexity, instead of jumping directly to the super-complex estimation scheme. 
 * A unified estimator is really what we want? While the "superestimator" is quite elegant, for practical reasons we may discover that is more convenient to use two different estimators (for example the "dynamics" one for the torque estimation and the "kinematics" one for floating base estimation). We should be ready to adapt our infrastructure to what better fits our need in reality, because once we move to the real robot is probable that a lot of our assumption will be proven wrong. 

In the next days I will try to add some more documents on what we need to implement BERDY, going more in the technical details. 

Comments from people involved in a way or another in estimation are more then welcome: 
@iron76 @naveenoid @jeljaik @nunoguedelha @claudia-lat @prashanthr05

iron76, 2015-07-17:
Thanks Silvio for bringing this up. 

1. Concerning the floating base estimation, I think that we need some very basic module which at least estimates the orientation, starting from the estimation of the Xsens sensor. I think this is really needed by @DanielePucci. Then we can progressively move towards better estimators once we can do a better job than Xsens. This is also necessary for comparison.

2. I really like agree the road to BERDY. What may be is missing is the hyper-parameters estimation which is also fundamental to make the approach effective.

DanielePucci, 2015-07-18:
From the control perspective, the looming tasks of balancing on dynamic surfaces as well as walking cannot be performed efficiently with an estimation of the floating base that depends only on the joint kinematics. In some cases, the estimation of the floating base w.r.t. an inertial frame is necessary for making the robot react, e.g. balancing on seesaw. Therefore, some kind of base estimation is, right now, necessary in view of the control tasks we have to achieve.  

I would start with the simplest solutions. For this reason, even before the EKF developed by @jeljaik, I would use the Xsens combined with forward kinematics - as @iron76 was mentioning - which seems to be the solution adopted by many teams working on the whole-body-control, and then proven to be effective.

naveenoid, 2015-07-18:
Few notes :
On a non-dynamic case, I tested balancing across a range of angles with
both the IMU and with Jor's EKF, i.e. with the current approach of using
inverse kinematics to floating base. Both sensor sources give stable
balancing results in the range of +- 10 degrees (when feet dont slip).
There is an issue with the IMU's output but I intend to test it
systematically next week. I will report back with the test results.

There is a technical issue before @DanielePucci and @Silvio 's suggestion
is applied. First, @jeljaik 's work on the EKF is so far only tested on the
accelerometer and gyroscope on the XSens, i.e. it must be physically
mounted on the surface. It might make sense to already move towards the
suggestion @iron76 made much earlier with Marco Maggiali to use the palm
skin board inside the foot to give access to athe accelerometer and
gyroscope on it (current board only has accelerometers). Maybe we can start
work with direct accelerometer and gyro fusion on this board.

About hyper parameter estimation, Im taking up that task, Ive started some
theoretical explorations regarding that. This however will first take place
using the matlab PNEA and DNEA. I can speed up this work asap if its
necessary. I was going to focus on the footstate estimation problem for
this, since its a basic rigid body before moving to the articulated case.
If someone wants to suggest a better roadmap for me on this, please let me
know.

on minor comment :
@Silvio, in the list in the codyco wiki we must start using the names PNEA
(T2) and DNEA (T3) for the respective phases to be consistent with the
algorithms mentioned in the publications.

---
Naveen Kuppuswamy, PhD
Post-doctoral Fellow,
Cognitive Humanoids Lab,
Department of Robotics, Brain and Cognitive Sciences (RBCS),
Istituto Italiano di Tecnologia, Genova, Italy

traversaro, 2015-07-18:
@naveenoid note that your email response get rendered on github as : 

~~~
Few notes :
…
~~~

Where then the complete reply is visible by clicking on `...` .

Replying in topic: 
 * I like the idea of using PNEA and DNEA acronyms, but I never found a definition of them in the papers, I just found them mentioned in the `bnt_time_varying` repo. 
 * For a using the XSens I think @DanielePucci intended to use the IMU that is already existing in the head of the robot. I think anyway it could make sense to add some scenarios to the roadmap to precisely define the things to implement.

DanielePucci, 2015-09-16:
Yes Silvio, I intended that. The definition of some scenario is also doable. As a first one, I may think of balancing on the seesaw, which is one of the next control problems we have to deal with. I also think, however, that this definition requires some brainstorming/official meeting among us, which may define what we have to face next @iron76.

traversaro, 2018-04-18:
The BERDY algorithms was implemented successfully for humans (see https://github.com/robotology/human-dynamics-estimation) and robots (see https://github.com/robotology/codyco-modules/pull/240). I think we can close this issue.

closed 2018-04-18 by traversaro