robotology-legacy/codyco-superbuild #83: [doc] port matlab installation instructions in the main superbuild readme

traversaro, 2015-09-25:
In the WBI-Toolbox README.md there are some useful information that are appliable to all matlab-based superbuild projects  (such as adding the addpath([getenv(CODYCO_SUPERBUILD_DIR)  /install/mex) directory to the matlab path).
We should bring relevant information in the main superbuild readme, so it can be used also by users that just want to use the iDynTree matlab bindings or the mex-wholebodymodel.

traversaro, 2015-10-21:
At least for Linux and OS X, perhaps we can just tell the users to set the appropriate `MATLABPATH` environmental variable on their `.bashrc` ? 
This will be consistent with all the other things that we set in the .bashrc.. @jeljaik @francesco-romano ?

francesco-romano, 2015-11-03:
If matlab is launched from the UI the environmental variables will not be read

traversaro, 2015-11-03:
Right.

traversaro, 2016-02-25:
We should also include information on how to make sure that CMake finds matlab, as done in https://github.com/robotology/WB-Toolbox#compiling-the-c-code-mex-file .

> Prerequisite: Check the Matlab configuration. Before going ahead with the compilation of the library, make sure that you have MATLAB and Simulink properly installed and running. Then, check that the MEX compiler for MATLAB is setup and working. For this you can try compiling some of the MATLAB C code examples as described in [http://www.mathworks.com/help/matlab/ref/mex.html#btz1tb5-12]. If you installed Matlab in a location different from the default one, please set an environmental variable called either MATLABDIR or MATLAB_DIR with the root of your Matlab installation, e.g. add a line to your ~/.bashrc such as: export MATLAB_DIR=/usr/local/bin/matlab

traversaro, 2016-08-03:
A convenient facility provided by `WB-Toolbox` is the [`startup_wbitoolbox.m`](https://github.com/robotology/WB-Toolbox/blob/master/toolbox/.startup_wbitoolbox.m.in) script automatically generated when the `WB-Toolbox` cmake is run. 

@gabrielenava wanted to introduce a similar mechanism in https://github.com/robotology/mex-wholebodymodel/pull/71 , but I think it is convenient to add it in the superbuild (as `startup_codyco_superbuild.m`) so all projects can benefit from it. 

We volunteered to investigate the issue.

MichaelBombile, 2017-04-18:
Hi , we are trying to implement a reactive walking controller on icub2 at EPFL LASA, and to do so we are trying to combine it with the matlab torqueBalancing module by exploiting the ComDes:i and the qDes:i input ports. However in our implementation we tried to add an inverse Kinematics box from the WBT_toolbox in order to compute the qDes and it keeps on giving the following error msg: Error:Error reported by S-function 'WBToolbox' in 'torqueBalancingAndWalking/Inverse Kinematics1/S-Function':
    [mdlStart] Cannot retrieve string from robot part parameter.

Here are the configuration parameters during the compilation:

[INFO][robotStatus::robotConfig] Configuring...
    [INFO][robotStatus::robotConfig] wbiConfFileFullPath is: /home/michael/icub/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini
    [INFO][robotStatus::robotConfig] List of properties for yarpWholeBodyInterface was interpreted correctly
    [INFO][robotStatus::robotConfig] robot_fixed is false!
    [DEBUG][robotStatus::robotConfig] After reading from config file, params are 
    [DEBUG][robotStatus::robotConfig] robot name:            icubGazeboSim 
    [DEBUG][robotStatus::robotConfig] local name:            matlabTorqueBalancing 
    [DEBUG][robotStatus::robotConfig] world reference frame: l_sole 
    [DEBUG][robotStatus::robotConfig] WBI ID List:           ROBOT_TORQUE_CONTROL_JOINTS_WITHOUT_PRONOSUP

Please, can anyone tell me what is wrong and how we can fix it?

traversaro, 2017-04-18:
Hi @MichaelBombile , your problem seems to be different from this issue, can you open a separate issue in a related repository, for example https://github.com/robotology-playground/WBI-Toolbox-controllers/issues/new ? 
Thanks a lot.

MichaelBombile, 2017-04-18:
@traversaro  Thanks for your response. I will do that.

gabrielenava, 2017-09-07:
I think at this point the [file](https://github.com/robotology/codyco-superbuild/blob/master/.startup_codyco_superbuild.m.in) that I added to comply to the requirements of this issue is outdated, because `mex-wholebodymodel` is going to be removed from the superbuild. We should think about removing this file from the superbuild.

traversaro, 2018-04-19:
Spiritual successor: https://github.com/robotology/robotology-superbuild/issues/38 .

closed 2018-04-19 by traversaro