robotology-legacy/codyco-superbuild #86: Make errors on orocos_kdl

hu-yue, 2015-10-15:
Hi,
I updated with make-update all and got errors on orocos_kdl, then I deleted the build and tried again, this did not work, and also just freshly cloning the repo did not work out.
This is the error:
```
-- Configuring done
loading initial cache file /home/yue/iCubSW/codyco-superbuild/build/main/comStepper/CMakeFiles/YCMTmp/comStepper-cache.cmake
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yue/iCubSW/codyco-superbuild/build/libraries/wholeBodyInterface
-- Generating done
-- Build files have been written to: /home/yue/iCubSW/codyco-superbuild/build/misc/sensorsInterfaces
[ 10%] -- Using iCub from build tree
-- Generating done
[ 11%] -- Build files have been written to: /home/yue/iCubSW/codyco-superbuild/build/external/qpOASES
-- Found YCM: /usr/local/share/YCM (found version "0.2.0~20151008.6+gitdb6a7b4")
Performing build step for 'orocos_kdl'
Performing build step for 'InSituFTCalibration'
-- Configuring done
make[3]: *** No targets specified and no makefile found.  Stop.
-- Configuring done
make[2]: *** [external/orocos_kdl/CMakeFiles/YCMStamp/orocos_kdl-build] Error 2
make[1]: *** [CMakeFiles/orocos_kdl.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

The full log from make and the CMakeCache are here: https://www.dropbox.com/sh/0mgj2fqk63l18h3/AAA0SMZS4DMxZvG46Nl7eJ7Fa?dl=0
icub-main, yarp and ycm were independently installed and updated today.
Thanks!

traversaro, 2015-10-15:
Duplicate of https://github.com/robotology/codyco-superbuild/issues/72 and https://github.com/robotology/ycm/issues/81 . 

Basically there is a problem with master YCM in compiling the codyco-superbuild . We solved it in the superbuild by downloading YCM v0.1.1 instead of the master version. 

Possible workarounds if you have an external YCM is to use YCM v0.1.1 , by doing `git checkout v0.1.1` .

 We should mention this in the readme.

hu-yue, 2015-10-15:
Ok thanks, didn't see these issues.
This solved the problem, yes I think you should point it out, even if it was my fault not looking into the old issues.

closed 2015-10-15 by hu-yue

traversaro, 2015-10-15:
No, definitely necessary to point it out in the README, we cannot expect all the users to read all the old issues. :smile: