robotology-legacy/codyco-superbuild #88: Errors with IDynTree and mexWholeBodyModel

KevinAStein, 2015-10-22:
Hi,
when compiling I get the following error:
```
codyco-superbuild/main/mexWholeBodyModel/src/modelstate.cpp:25:45: fatal error: iDynTree/ModelIO/URDFDofsImport.h: No such file or directory
 #include <iDynTree/ModelIO/URDFDofsImport.h>
                                             ^
compilation terminated.
make[5]: *** [CMakeFiles/wholeBodyModel.dir/src/modelstate.cpp.o] Error 1
make[4]: *** [CMakeFiles/wholeBodyModel.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [main/mexWholeBodyModel/CMakeFiles/YCMStamp/mexWholeBodyModel-build] Error 2
make[1]: *** [CMakeFiles/mexWholeBodyModel.dir/all] Error 2
make: *** [all] Error 2
```
@hu-yue

traversaro, 2015-10-22:
Please update the superbuild with `make update-all`, thanks.

traversaro, 2015-10-22:
@hu-yue by the way, are you  using mex-wholebodymodel ? We are changing a lot of its API in these days.

KevinAStein, 2015-10-22:
@traversaro for the time being we are not using it

traversaro, 2015-10-22:
Thanks for the info! The update-all solved the compilation issue?

KevinAStein, 2015-10-22:
yes, its compiling now.

closed 2015-10-22 by KevinAStein