robotology-legacy/codyco-superbuild #89: codyco-superbuild failing after latest change in icub-main

traversaro, 2015-10-23:
The nightly compilation of the codyco-superbuild [1] is now failing due to the latest changes on icub-main [2].

 Anyone willing to fix it? @robotology/codyco-developers 

[1] : https://travis-ci.org/robotology/codyco-superbuild/builds/86923104
[2] : https://github.com/robotology/QA/issues/68

jeljaik, 2015-10-23:
If it can wait a couple of hours till I get home I can do it.

jeljaik, 2015-10-23:
Fixed in https://github.com/robotology/codyco-modules/commit/591248dc12def5255f9fd673d823fb39f2a5d21e

traversaro, 2015-10-24:
Travis is now happy: https://travis-ci.org/robotology/codyco-superbuild/builds/87128502 , closing.

closed 2015-10-24 by traversaro

jeljaik, 2015-10-25:
If I ever get a robot at home, I'll call him Travis.