robotology-legacy/codyco-superbuild #92: RPATH related issue on MacOS

iron76, 2015-10-30:
I am getting the following error when launching either the `wholeBodyDyanmicsTree` or the `torqueBalacing`:
```
dyld: Library not loaded: @rpath/libyarpwholebodyinterface.0.2.3.dylib
```
Adding the definitions:
``` 
export DYLD_LIBRARY_PATH=$DYLD_LIBRARY_PATH:$CODYCO_SUPERBUILD_ROOT/build/install/lib
```
was not sufficient since it lead to following error:
```
dyld: Library not loaded: @rpath/libYARP_math.1.dylib
```
Solution was also to define:
```
export DYLD_LIBRARY_PATH=$DYLD_LIBRARY_PATH:/path/to/yarp/lib
```
Not sure my configuration is correct but it was working before this commit https://github.com/robotology/yarp/commit/ac69d1b7d27d99c274d849c21d40ee4aa4f36595 which might be related.

iron76, 2015-10-30:
The issue is propagating. Even if my MATLAB was properly configured in the past, it seems that now the same library loading problem has been propagated to the `WBI-toolbox` configuration. I am getting the following error:
```
Caused by:
Invalid MEX-file '[...]codyco-superbuild/build/install/mex/yarpRead.mexmaci64': dlopen([...]codyco-superbuild/build/install/mex/yarpRead.mexmaci64, 1): Library not loaded: @rpath/libYARP_math.1.dylib
Referenced from: [...]codyco-superbuild/build/install/mex/yarpRead.mexmaci64
Reason: image not found
```

iron76, 2015-10-30:
The problem was solved as suggested in the installation instructions configuring .matlab7rc.sh to define `DYLD_LIBRARY_PATH` as defined in the system environment variable.

iron76, 2015-10-30:
Additional problems with the MacOSX10.11.sdk were solved by launching again the `mex -setup` in MATLAB.

francesco-romano, 2015-10-30:
Unfortunately the use of `DYLD_LIBRARY_PATH` is not a solution but a workaround.
As soon as you have some time you should try again with clean builds starting with yarp and iCub (if you are able to launch `yarpserver` and/or some iCub app then you should be ok.

iron76, 2015-11-01:
It was indeed a fresh install. YARP and iCub applications work fine. I will try another fresh install as you recommend.

iron76, 2015-11-01:
Fresh install (removing all CMakeCache and all binaries) solved the issue.

closed 2015-11-01 by iron76