robotology-legacy/codyco-superbuild #94: Headers not updated when changes are made

rlober, 2015-11-19:
We have recently been making some major changes to `ocra-wbi-plugins` and I am encountering a new issue with CMake that I wasn't having before. Basically whenever I make changes to a header file in our source code of the shared libs, those changes aren't recognized and the header referenced from `build/install/{project}/{header}.h` never changes. It seems like CMake is ignoring the header dependencies, but I cannot seem to figure out how to fix it in our CMakes and I am wondering if we are missing some codyco option for the build. The executables we build `wocraController` for example work just fine and whenever I make changes to a header it is updated in the `install/include` dir.

traversaro, 2015-11-19:
It seems an issue on on the order of inclusion directories. 

While compiling your library should be always using the headers from the sources directory, never the one from the installed directory. 

Given that the order of how the include directories are searched depends is the one in which the directories are passed to the compiler (at least for GCC [1]) and this order is determined by the order in which the directories are passed to the `include_directories` cmake command [2], I guess in your case some early `include_directories`of a system library (such as https://github.com/ocra-recipes/ocra-wbi-plugins/blob/master/modules/ocraObservers/CMakeLists.txt#L37 ? ) could add the superbuild install headers directory **before** the source one, shadowing the source include directories that should be used. 

I hope this could be solved by: 
  * Always put the `INCLUDE_DIRECTORIES(${PROJECT_SOURCE_DIR}/include` as the first include directories command (or put the equivalent `target_include_directories(targetName ${PROJECT_SOURCE_DIR}/include`command as the first include command)
  * Include the directories of headers from third party libraries using `include_directories(SYSTEM ${Eigen_INCLUDE_DIRS} ${EigenLgsm_INCLUDE_DIR})`. At least in GCC [1] the system include directory have lower precedence with respect to normal ones. 

[1] : https://gcc.gnu.org/onlinedocs/cpp/Search-Path.html
[2] : https://cmake.org/cmake/help/v3.0/command/include_directories.html

rlober, 2015-11-19:
Thanks for the help @traversaro.
cc: @ahoarau
You were definitely on track with the include directory order. It took a little bit of google foo but I was able to write this little snippet to see the include dir paths for the projects:

```cmake
GET_DIRECTORY_PROPERTY(inc_dirs INCLUDE_DIRECTORIES)
message(
"
Project include dirs:
${inc_dirs}
"
)
```

This printed out all the include directory paths and sure enough `codyco-superbuild/build/install/include` was of higher precedence than `codyco-superbuild/main/ocra-wbi-plugins/.../include` basically meaning the real source headers were being ignored. This include dir came from the line: https://github.com/ocra-recipes/ocra-wbi-plugins/blob/master/CMakeLists.txt#L29

where I include `${yarpWholeBodyInterface_INCLUDE_DIRS}` which apparently isn't even necessary given the structure of superbuild.

Anyways, thanks for the pointer :+1: and btw: we are now compatible with Eigen 3.2.0! Yay for @ahoarau.

closed 2015-11-19 by rlober

traversaro, 2015-11-19:
:+1: :+1: :+1: For the Eigen 3.2 news. 
By the way, do you still have machines on Ubuntu 12.04?

rlober, 2015-11-19:
Nope! We are now in 2014 with 14.04 and Gazebo 6! We need to work out one or two usage bugs and then you guys should be able to run all of our stuff without any dependency conflicts - if you want.

Also, are you going to Birmingham for the meeting?

traversaro, 2015-11-19:
That is great! 

I am currently in Netherlands until Christmas, and I don't remember if I am supposed to go to the mid-year meeting.

rlober, 2015-11-19:
What are you doing there?

traversaro, 2015-11-19:
I am visiting [TU/e](https://www.tue.nl/en/) for three months.

rlober, 2015-11-19:
Ahhhh well enjoy!

DanielePucci, 2015-11-20:
Hey @rlober, I am coming to Birmingham, and I think that our beloved collaborator @traversaro is supposed to come. Am I right? @iron76

rlober, 2015-11-20:
Yeah man! It will be another 24h codyco party!

DanielePucci, 2015-11-20:
Indeed!!!

DanielePucci, 2015-11-20:
@vpadois

vpadois, 2015-11-20:
@traversaro @rlober @DanielePucci @iron76 : an iCub party without an iCub is not really an iCub party. But I will be there anyway (at least to take care of Junior).