robotology-legacy/codyco-superbuild #100: Problems compiling codyco-superbuild, errors on orocos_kdl and qpOASES

hu-yue, 2015-12-18:
Hi,
I had qpOASES installed from external sources (not from superbuild), then I could not compile supebuild since it was complaining that it could not find qpOASES, then I uninstalled the qpOASES of the external sources and tried again with codyco-superbuild, it kept having errors. Given the frustration, I decided to delete the whole codyco-superbuild and build everything again, and these are the errors I get:
 1. Cmake complains ```ERRORCould not find qpOASES```
 2. I decided to proceed despite error n. 1 and got the following errors
```
[  2%] [  2%] [  4%] [  5%] Skipping update step for 'paramHelp'
Skipping update step for 'orocos_kdl'
Skipping update step for 'wholeBodyInterface'
Skipping update step for 'qpOASES'
[  6%] [  8%] [  9%] [ 11%] Performing configure step for 'qpOASES'
Performing configure step for 'orocos_kdl'
Performing configure step for 'paramHelp'
Not searching for unused variables given on the command line.
Performing configure step for 'wholeBodyInterface'
Not searching for unused variables given on the command line.
CMake Error: The source directory "/home/yue/iCubSW/codyco-superbuild/external/orocos_kdl" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
Not searching for unused variables given on the command line.
Not searching for unused variables given on the command line.
make[2]: *** [external/orocos_kdl/CMakeFiles/YCMStamp/orocos_kdl-configure] Error 1
make[1]: *** [CMakeFiles/orocos_kdl.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
-- Configuring done
-- Found YCM: /usr/local/share/YCM (found version "0.1.1")
-- wholeBodyInterface version 0.2.3
-- Found YCM: /usr/local/share/YCM (found version "0.1.1")
-- Generating done
-- Build files have been written to: /home/yue/iCubSW/codyco-superbuild/build/external/qpOASES
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yue/iCubSW/codyco-superbuild/build/libraries/wholeBodyInterface
-- Configuring done
[ 12%] -- Generating done
Performing build step for 'qpOASES'
-- Build files have been written to: /home/yue/iCubSW/codyco-superbuild/build/libraries/paramHelp
[ 13%] Performing build step for 'wholeBodyInterface'
[ 15%] Performing build step for 'paramHelp'
[100%] Built target qpOASES
[100%] Built target whole-body-interface
[ 16%] Performing install step for 'qpOASES'
[100%] Built target paramhelp
[ 18%] Performing install step for 'wholeBodyInterface'
[ 19%] Performing install step for 'paramHelp'
[100%] Built target qpOASES
Install the project...
[100%] Built target whole-body-interface
-- Install configuration: "Release"
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/lib/libqpOASES.so.3.1
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/lib/libqpOASES.so
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES.hpp
-- Installing: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES
-- Installing: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/extras
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/extras/SolutionAnalysis.hpp
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/extras/SolutionAnalysis.ipp
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/extras/OQPinterface.hpp
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/Bounds.ipp
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/ConstraintProduct.hpp
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/Bounds.hpp
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/Options.hpp
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/SQProblem.hpp
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/Types.hpp
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/QProblemB.hpp
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/QProblemB.ipp
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/SubjectTo.hpp
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/SubjectTo.ipp
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/Matrices.hpp
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/Indexlist.ipp
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/SQProblem.ipp
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/Constraints.ipp
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/qpOASES/MessageHandling.hpp
[... 46 lines cut ...]
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/lib/libparamhelp.so.0.0.3
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/lib/libparamhelp.so
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/paramHelp/paramHelpUtil.h
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/paramHelp/paramProxyInterface.h
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/paramHelp/paramProxyBasic.h
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/paramHelp/paramHelperBase.h
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/paramHelp/paramHelperClient.h
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/include/paramHelp/paramHelperServer.h
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/lib/cmake/paramHelp/paramHelpConfigVersion.cmake
-- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/lib/cmake/paramHelp/paramHelpConfig.cmake
[ 22%] -- Up-to-date: /home/yue/iCubSW/codyco-superbuild/build/install/lib/cmake/paramHelp/paramHelpTargets.cmake
-- Installing: /home/yue/iCubSW/codyco-superbuild/build/install/lib/cmake/paramHelp/paramHelpTargets-release.cmake
Completed 'wholeBodyInterface'
[ 23%] [ 27%] Built target qpOASES
Completed 'paramHelp'
[ 31%] Built target wholeBodyInterface
[ 36%] Built target paramHelp
make: *** [all] Error 2
```

I already checked as for issue #86 and the ycm version is now still ```v0.1.1```, not sure if I should switch it back to master.
All started with a segfault in a program I wrote some time ago using yarpWholeBodyInterface, trying to debug that on a computer where I did not dare to update the superbuild.
CMakeCache is here https://www.dropbox.com/s/jn91au2nr3kzndg/CMakeCache.txt?dl=0

Thanks!

traversaro, 2015-12-18:
With "I did not dare to update the superbuild" you mean you did not run neither `git pull` or `make update-all`? In this case what is the latest commit on your local codyco-superbuild repo (use `git log -n 1` to discover it)?

hu-yue, 2015-12-18:
Yes exactly I did not perform any pull or update-all, this is the latest commit fa6d240ca00b417426d1e04b0cf3bbe03c7e09f7

hu-yue, 2015-12-18:
But I also did not install qpOASES from external sources on that computer, not sure if this matters somehow...

hu-yue, 2015-12-18:
And the segfault problem I was talking about before is not there anymore, but I still did not update the superbuild.

traversaro, 2015-12-18:
I think the problem is the YCM is not at the master, because the compilation of `orocos_kdl` depends on the `CONFIGURE_SOURCE_DIR`, that is available only on ycm `master`. [1] 
Until a new version of ycm is released (@drdanz happening any soon?) for checking that the version of ycm available is compatible with the `Buildorocos_kdl.cmake` we should add the check explained in https://github.com/robotology/ycm/issues/81#issuecomment-152988244 . 

[1] https://github.com/robotology/codyco-superbuild/blob/master/cmake/Buildorocos_kdl.cmake#L10

hu-yue, 2015-12-18:
@traversaro thanks, I switched ycm to master now and it seems to fix the orocos_kdl problem.
However if I have qpOASES installed from external sources it keeps saying that qpOASES cannot be found, but still compiles without problems.
 ```
CMake Warning at /usr/local/share/YCM/modules/FindOrBuildPackage.cmake:283 (find_package):
   Could not find a package configuration file provided by "qpOASES" with any
   of the following names:

     qpOASESConfig.cmake
     qpoases-config.cmake

   Add the installation prefix of "qpOASES" to CMAKE_PREFIX_PATH or set
   "qpOASES_DIR" to a directory containing one of the above files.  If
   "qpOASES" provides a separate development package or SDK, be sure it has
   been installed.
 Call Stack (most recent call first):
   CMakeLists.txt:139 (find_or_build_package)
 ```
Also, this error message with qpOASES not found ```ERRORCould not find qpOASES``` when it is not installed from external sources is somehow misleading as well.

traversaro, 2015-12-23:
Where is installed the external qpOASES ? 
At the moment qpOASES is just used in the simulink controllers I think.

hu-yue, 2015-12-25:
The external qpOASES is in  ```/usr/local ```, the source code is from ```https://projects.coin-or.org/svn/qpOASES/trunk```

traversaro, 2015-12-31:
qpOASES does not install any `**Config.cmake` configuration files for find the library with a `find_package(qpOASES)` command:
~~~
pegua@pareto:~/src/qpOASES-3.2.0/build$ make install
[ 62%] Built target qpOASES
[ 65%] Built target example1
[ 68%] Built target example1a
[ 72%] Built target example1b
[ 75%] Built target example2
[ 79%] Built target example3
[ 82%] Built target example3b
[ 86%] Built target example4
[ 89%] Built target example5
[ 93%] Built target exampleLP
[ 96%] Built target qrecipe
[100%] Built target qrecipeSchur
Install the project...
-- Install configuration: "Release"
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/lib/libqpOASES.a
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/ConstraintProduct.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/Utils.ipp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/Types.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/extras
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/extras/SolutionAnalysis.ipp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/extras/SolutionAnalysis.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/extras/OQPinterface.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/Flipper.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/SQProblemSchur.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/QProblemB.ipp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/MessageHandling.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/Utils.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/SQProblem.ipp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/MessageHandling.ipp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/QProblemB.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/Options.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/SubjectTo.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/Matrices.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/Indexlist.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/QProblem.ipp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/UnitTesting.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/SparseSolver.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/QProblem.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/SubjectTo.ipp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/Constraints.ipp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/SQProblem.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/SQProblemSchur.ipp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/Bounds.ipp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/Bounds.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/Constants.hpp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/Indexlist.ipp
-- Installing: /home/pegua/src/qpOASES-3.2.0/build/./install/include/qpOASES/Constraints.hpp
~~~

That's actually why we keep a separate repo with a decent "CMake" build system for qpOASES at https://github.com/robotology-dependencies/qpOASES (this is the one used by the superbuild). 

Porting this changes upstream is desirable but we never had the time to do it, see https://github.com/robotology-dependencies/qpOASES/issues/1 . 

Even if qpOASES does not install any CMake configuration file, qpOASES should be found by https://github.com/robotology/ycm/blob/master/find-modules/FindqpOASES.cmake , I wonder why it is not found. I will try to reproduce it locally.

francesco-romano, 2016-11-09:
Hi @hu-yue 
Considering that you successfully tested the balancing on your version of the iCub robot, I assume that this issue has been solved.
If not, feel free to open it again.

closed 2016-11-09 by francesco-romano