robotology-legacy/codyco-superbuild #98: Testing controller performance automatically.

rlober, 2015-12-18:
As I mentioned in the commit d23a6e5c83fa394de591103599aba82e0ddacaae I think we need a way of testing controller performance automatically when updates have been made to it. I know we have build testing but I am too unfamiliar with Travis to know if we can run executables as well. On our end (@ahoarau) we can most definitely create a test file which could run some specific set of commands and check computation time. We could then check computation time average against some base truth.

traversaro, 2015-12-18:
With performance you mean time performance, right? 

For running tests that involve several different processes (and can be run both on a simulator or on a real robot) at IIT we developed the [Robot Testing Framework](https://github.com/robotology/robot-testing), but unfortunately I started sometime ago to write tests involving gazebo, but I still did not finish, see https://github.com/robotology/gazebo-yarp-plugins/pull/194 . 

On our side I can work on providing a test example using Gazebo,  but not on the short term unfortunately (for sure after Christmas). 

An alternative approach would be to have a test that run the controller with a "simple" model (for example https://github.com/ocra-recipes/gOcra/tree/master/examples/test_GHCJT) a then run that on a test executable and benchmark its execution time. This is easier, but perhaps too many factors affecting performances (ocra-wbi interface, the high number of DOFs) would be untested.

rlober, 2015-12-18:
Yeah, time performance.

No worries for the moment but I think it will be a crucial feature in the future when people start adding to the controller. I guess I can close this then because it is a really big topic and not quite an issue.

closed 2015-12-18 by rlober