robotology-legacy/codyco-superbuild #99: ccmake: Lua: disabled

misaki43, 2015-12-18:
Hi,
The problem is when I want to do:
 "cd $YARP_ROOT/bindings
mkdir build
cd build
ccmake .." 
and I want to put CREATE_LUA=ON, this "create_lua" variable doesn't exist. 
If I do "cmake .. -DCREATE_LUA=ON" it is written: "+++ Lua: disabled" and other disabled variables.
--------Some other information------------------
I have installed LUA with: "http://www.yarp.it/yarp_swig.html#yarp_swig_lua" 
and I don't really understand this part : 
"After Compiling YARP language bindings, make sure that the Lua-Yarp binding shared library (.so/.dll/.dylib/...) is included in your LUA_CPATH environment variable; e.g., on linux:
export LUA_CPATH=";;;/path/to/bindings/build/?.so""
So I try to put in the bashrc things like:
#lua path
export LUA_PATH=";;;/home/odermy/yarp/bindings/build/rFSM/?.lua"
export LUA_CPATH=";;;/:/usr/lib/x86_64-linux-gnu/?.so"
---------------------------------------------------------
Can you help me please?

traversaro, 2015-12-18:
Hi @misaki43 . 

Currently Lua is an optional dependency of the codyco-superbuild, used only for some (undocumented and currently not working) demos. Depending on your goals, I think you can safely skip it.

traversaro, 2015-12-18:
I removed that section from the documentation, because Lua is not working in any *maintained* software/demo, see https://github.com/robotology/codyco-superbuild/commit/e423d21cf9502ffb859345129b7d58489803786c . Thanks for the issue!

traversaro, 2015-12-23:
@misaki43 The installation went ok I assume?

misaki43, 2015-12-28:
@traversaro : Yes the installation is ok. What does Lua serve if it is optional?

traversaro, 2016-01-13:
It was necessary for some demos (in particular for running some finite state machine using [rFSM](https://github.com/kmarkus/rFSM) ) but this demos are not running at the moment.

traversaro, 2016-01-13:
I will close the issue, let me know if you have any further question. : )

closed 2016-01-13 by traversaro