robotology-legacy/codyco-superbuild #106: Build Error on osx El Capitan (whole source)

ahoarau, 2016-02-10:
Don't know if I'm supposed to, but I tried to compile the superbuild including yarp and iCub and I get : 
```CompileC /Users/Antoine/Desktop/codyco-superbuild/build/external/YARP/extern/goocanvas/YARP.build/Release/YARP_priv_goocanvas.build/Objects-normal/x86_64/goocanvas.o extern/goocanvas/goocanvas/src/goocanvas.c normal x86_64 c com.apple.compilers.llvm.clang.1_0.compiler
    cd /Users/Antoine/Desktop/codyco-superbuild/external/YARP
    export LANG=en_US.US-ASCII
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -x c -arch x86_64 -fmessage-length=0 -fdiagnostics-show-note-include-stack -fmacro-backtrace-limit=0 -Wno-trigraphs -fpascal-strings -O3 -Wno-missing-field-initializers -Wno-missing-prototypes -Wno-return-type -Wno-missing-braces -Wparentheses -Wswitch -Wno-unused-function -Wno-unused-label -Wno-unused-parameter -Wno-unused-variable -Wunused-value -Wno-empty-body -Wno-uninitialized -Wno-unknown-pragmas -Wno-shadow -Wno-four-char-constants -Wno-conversion -Wno-constant-conversion -Wno-int-conversion -Wno-bool-conversion -Wno-enum-conversion -Wno-shorten-64-to-32 -Wpointer-sign -Wno-newline-eof -DCMAKE_INTDIR=\"Release\" -DYARP_PRESENT -D_REENTRANT -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk -fasm-blocks -fstrict-aliasing -Wdeprecated-declarations -mmacosx-version-min=10.11 -Wno-sign-conversion -I/Users/Antoine/Desktop/codyco-superbuild/build/external/YARP/lib/Release/include -I/Users/Antoine/Desktop/codyco-superbuild/external/YARP/extern/goocanvas/goocanvas -I/Users/Antoine/Desktop/codyco-superbuild/external/YARP/extern/goocanvas/goocanvas/src -I/usr/local/include/atk-1.0 -I/usr/local/include/cairo -I/usr/local/include -I/usr/local/lib/gtk-2.0/include -I/usr/local/include/gtk-2.0 -I/usr/local/include/gdk-pixbuf-2.0 -I/usr/local/lib/glib-2.0/include -I/usr/local/include/glib-2.0 -I/usr/local/include/pango-1.0 -I/usr/local/include/freetype2 -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/System/Library/Frameworks/Tk.framework/Headers/X11 -I/Users/Antoine/Desktop/codyco-superbuild/build/external/YARP/extern/goocanvas/YARP.build/Release/YARP_priv_goocanvas.build/DerivedSources/x86_64 -I/Users/Antoine/Desktop/codyco-superbuild/build/external/YARP/extern/goocanvas/YARP.build/Release/YARP_priv_goocanvas.build/DerivedSources -Wmost -Wno-four-char-constants -Wno-unknown-pragmas -F/Users/Antoine/Desktop/codyco-superbuild/build/external/YARP/lib/Release -DNDEBUG -fPIC -MMD -MT dependencies -MF /Users/Antoine/Desktop/codyco-superbuild/build/external/YARP/extern/goocanvas/YARP.build/Release/YARP_priv_goocanvas.build/Objects-normal/x86_64/goocanvas.d --serialize-diagnostics /Users/Antoine/Desktop/codyco-superbuild/build/external/YARP/extern/goocanvas/YARP.build/Release/YARP_priv_goocanvas.build/Objects-normal/x86_64/goocanvas.dia -c /Users/Antoine/Desktop/codyco-superbuild/external/YARP/extern/goocanvas/goocanvas/src/goocanvas.c -o /Users/Antoine/Desktop/codyco-superbuild/build/external/YARP/extern/goocanvas/YARP.build/Release/YARP_priv_goocanvas.build/Objects-normal/x86_64/goocanvas.o
In file included from /Users/Antoine/Desktop/codyco-superbuild/external/YARP/extern/goocanvas/goocanvas/src/goocanvas.c:98:
/usr/local/include/glib-2.0/glib/gi18n-lib.h:23:10: fatal error: 'libintl.h' file not found
#include <libintl.h>
         ^
1 error generated.```

kjyv, 2016-02-10:
It seems you need to install gettext, possibly using homebew or macports

ahoarau, 2016-02-10:
I brewed it , but it showed a warning like 'I might shadow the system gettext , don't forget to link against /usr/local/blabla '

kjyv, 2016-02-10:
I'm not sure about homebrew, I'm not using it. It might not have made the headers visible system wide.

ahoarau, 2016-02-11:
Still compiling but it seems ok after 
```
brew link gettext --force
```

traversaro, 2016-02-11:
I talked with @francesco-romano , and apparently `gettext` is necessary only when compiling GTK guis. Most of the YARP GTK guis have been deprecated and replaced by QT5 guis, except for some GUIs that are either not so used (`yarpbuilder`) or useful only on the pc104 (`canLoader` and `ethLoader`). 
With https://github.com/robotology/codyco-superbuild/commit/667de4aad22242e2ec8418faf27ae857111e6d86 I disabled (at least in default build) the compilation of YARPBUILDER in yarp compiled by the superbuild, and that should solve the problem even without having to run `brew link gettext --force`. 

More in general, installing yarp and icub-main directly using the superbuild was initially not mentioned in the superbuild  documentation because the typical setup at IIT is to compile yarp and icub-main separetly, but for external users it is definitly much more convenient. We whould add proper mention of this in the installation guide.

traversaro, 2016-02-16:
I consider this fixed by https://github.com/robotology/codyco-superbuild/commit/667de4aad22242e2ec8418faf27ae857111e6d86 , feel free to reopen if necessary.

closed 2016-02-16 by traversaro