robotology-legacy/codyco-superbuild #107: Problem with matlab

traversaro, 2016-02-12:
> Hi, i am having problem with MATLAB simulations. I am getting error like:
failed assertion in b:\matlab\toolbox\stateflow\src\stateflow\utils\object.cpp at line 2853:
object[id]!=NULL

> I looked into the post dated 29th Jan 2015 Memory allocation error with holder block #69
> but dont understand how to rectify this.

> Please help!

Hi @garimaiitd, can you provide more information on your problem. 
In particular it would be nice to know:
 * Operating System
 * Matlab Version
 * Which Matlab code or Simulink model are you using
Thanks!

garimaiitd, 2016-02-12:
- Operating System: WINDOWS 7
   - Matlab Version: MATLAB R2013b
   - Which Matlab code or Simulink model are you using: I am trying to
   simulate one simevents model given on the Mathworks site "
   http://blogs.mathworks.com/simulink/2015/02/04/surprising-results-from-modeling-queues-with-simevents/
   "

Thanks.

-Garima

francesco-romano, 2016-02-12:
Hi @garimaiitd 

If I understood correctly you are trying to simulate a model provided by Mathworks.
I don't know how this can be related to the software provided by us.

I think you should ask directly to Mathwork support. I doubt we can provide any.

garimaiitd, 2016-02-12:
ok. Thanks for your time and prompt response .

closed 2016-02-12 by francesco-romano