robotology-legacy/codyco-superbuild #109: Write documentation on how to modify a subproject in the superbuild

traversaro, 2016-02-18:
Step that should be documented: 
  * Open a subproject in an IDE (Qt-Creator, KDevelop, Visual Studio, Xcode) 
  * Add a simple main ("hello word") to an existing subproject 

@liesrock I remember something similar in walkman-superbuild documentation (opening a subproject in qt creator, etc etc) but I am not able to find it anymore. Any pointers? 

This should be added in the YCM documentation to avoid duplication, and properly linked in the `codyco-superbuild` readme. @drdanz I remember there were some stubs related to this (using superbuild with IDEs) in the YCM documentation, but I cannot find them. Any pointers?

traversaro, 2016-02-18:
cc @fjandrad

traversaro, 2016-02-18:
IDE stub: https://robotology.github.io/ycm/gh-pages/master/manual/ycm-superbuild.7.html#ides (can be modified in https://github.com/robotology/ycm/blob/master/help/manual/ycm-superbuild.7.rst#L637 ).
walkman docs:  https://github.com/robotology-playground/robotology-superbuild/wiki/robotology-superbuild-handbook#super-build-workflow

liesrock, 2016-02-19:
Hi @traversaro ,
you found it, @arocchi  did it.

drdanz, 2016-02-19:
> @drdanz I remember there were some stubs related to this (using superbuild with IDEs) in the YCM documentation, but I cannot find them. Any pointers?

@traversaro you already found it... :+1:
It would be great to have that documentation to YCM instead of in each superbuild.

traversaro, 2018-04-19:
Spiritual successor: https://github.com/robotology/robotology-superbuild/issues/66 .

closed 2018-04-19 by traversaro