robotology-legacy/codyco-superbuild #110: C++11 Build

rlober, 2016-02-26:
Could we make an option that would enable the C++11 flag for all projects?
```cmake
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
```

traversaro, 2016-02-26:
The current way of passing options to the subprojects is through the `CMAKE_CACHE_ARGS` option of the [YCMEPHelper](http://robotology.github.io/ycm/gh-pages/master/module/YCMEPHelper.html) function. Adding something (not just replacing what is there) to the `CMAKE_CXX_FLAGS` option of each project it is not supported. 

What is the use case for this? 
In the idea world the required standard should be setted only by the project that uses it, and should be propagated to downstream projects by the use of [CMake compile features](https://cmake.org/cmake/help/v3.1/manual/cmake-compile-features.7.html).

rlober, 2016-02-26:
I am converting our code to use the C++11 standard, most specifically the stl smart pointers, and I want to see how the controller behaves when all of the libs used in the controller code are also compiled with C++11. From a convenience standpoint it would just be nice to globally set this flag rather than set it for each project. 

Theoretically compiling our project with C++11 should only be beneficial but I would like to be sure as the controller loop is time sensitive as you well know.

traversaro, 2016-02-26:
I see.. @drdanz any ideas?

drdanz, 2016-02-26:
Before adding a new option, I'd rather try
```
export CXXFLAGS=-std=c++11
```
before running `cmake` in the superbuild for the first time.
This should (theoretically, I didn't test it) enable the -std=c++11 automatically for all the subprojects without the need of adding an option.

rlober, 2016-03-01:
Worked like a charm! Thanks @drdanz 

Just for the info of future readers. This line should be put in the `.bashrc` file to set the environment variable `CXXFLAGS` which cmake looks for.

closed 2016-03-01 by rlober