robotology-legacy/codyco-superbuild #116: Tasks for the third year review meeting demo

DanielePucci, 2016-03-22:
The third year review meeting will include also an experimental validation of the designed controllers. To this purposes, we - as IIT guys - are planning to show the following demos:

1. iCub balancing on a soft carpet;
2. iCub balancing on seesaws.

For the success of the above validations, the following tasks have been identified (the responsible of each task is listed after the task itself):

T0) Generate the model for iCub Darmstadt01 @traversaro 

T1) Investigate quality of joint velocities from High Resolution Encoders @DanielePucci

T2) Tool for Offset / C.M. FT sensors calibration @fjandrad

T3) Tool for fine encoders offsets calibration from accelerometers @nunoguedelha @naveenoid 

T4) Use the IMUs attached to the feet and Base (waste)
Mechanical : where and how do we attach them? (contact: Tisi)
Software: what about the driver? (contact: Accame)
Tests @traversaro @francesco-romano @DanielePucci

T5) Generating joint references compatible with CoM trajectories @francesco-romano 

T6) WholeBodyDynamicsTree on PC104 @traversaro 

T7) Controller re-design @DanielePucci

T8) Carpet stiffness estimation

@robotology/codyco-developers @azadm @hu-yue @serena-ivaldi @robertocalandra @misaki43 @serena-ivaldi

traversaro, 2016-03-31:
CoDyCo Deliverable 5.3 (available at https://github.com/robotology-playground/codyco-deliverables/blob/master/D5.3/pdf/D5.3.pdf) contains some details on the demonstraiton scenario.

DanielePucci, 2016-04-22:
Task 2) assigned to @TeoKa and me.

DanielePucci, 2016-04-22:
Expected deadlines and status:

1. Done

2. 28th April: current status, 0%

3. 13th May: current status, 30%

4. 13th May: current status, 35%

5. Cancelled

6. 6th May: current status, 25%

7. 28th April: current status, 50% (but nobody should know it)

8. 13th May, current status, 50%

9. 13th May, current status, 50%

DanielePucci, 2016-04-24:
The task T8 has the following sub-tasks:
- [ ] general cleaning adding comments of YOGA++ code;
- [ ] avoid the use of two different QPs for finding the control solutions associated with the one-foot and two-feet standing phase;
- [ ] add QP solver to the `balancingControllerOnSeesaw`
- [ ] add a feature to `balancingControllerOnSeesaw` so that to use the robot's IMU to retrive the seesaw orientation and velocity
- [ ] add the continuity constraint of torques to  `balancingControllerOnSeesaw`
- [ ] add the continuity constraint of torques to  `balancingController`
- [ ] add control of seesaw orientation to `balancingControllerOnSeesaw`
- [ ] modify `balancingController` so that to take into account compliant contacts

francesco-romano, 2016-11-09:
Given that the 3rd year meeting took place some months ago, I close this issue.

closed 2016-11-09 by francesco-romano