robotology-legacy/codyco-superbuild #118: Document things to check on the real robot before performing whole body torque control experiments

traversaro, 2016-04-12:
The relevant document to fill is https://github.com/robotology/codyco-superbuild/blob/master/doc/icub-whole-body-torque-control-faqs.md . 

@misaki43 Can you take care of filling this document using the notes that you took? You can modify that document by  opening a Pull Request ( https://help.github.com/articles/using-pull-requests/ ) on this repo. 

The document uses the Markdown format (for a tutorial see here: http://www.markdowntutorial.com), and once you edit it (and the Pull Request is merged in the repo) all the users will see the document at http://wiki.icub.org/codyco/dox/html/index.html .

traversaro, 2016-04-12:
cc @serena-ivaldi @DanielePucci

traversaro, 2018-04-19:
Could it make sense to keep in `whole-body-controllers`?

gabrielenava, 2018-04-19:
Yes, in the [doc](https://github.com/robotology/whole-body-controllers/tree/master/doc) folder. I can move the document there.

traversaro, 2018-04-19:
Great, I think it can also address https://github.com/robotology/codyco-superbuild/issues/155 .

gabrielenava, 2018-04-19:
done with https://github.com/robotology/codyco-superbuild/commit/83704ec5c9898fa26c7ceb2d9380da29e4c9e52c

gabrielenava, 2018-05-28:
I think we can close this issue.

closed 2018-05-28 by gabrielenava