robotology-legacy/codyco-superbuild #126: Errors compiling urdfdom [OS X]

jeljaik, 2016-05-10:
I have just updated `codyco-superbuild` and deleted my `build` directory, yet when compiling the `urdfdom` project, it does not finish successfully with the following message: 

```bash
[...]
Performing configure step for 'urdfdom'
cd /Users/jorhabibeljaik/Code/codyco-superbuild/build/external/urdfdom && /usr/local/Cellar/cmake/3.5.2/bin/cmake --no-warn-unused-cli -DCMAKE_PREFIX_PATH:PATH=/Users/jorhabibeljaik/Code/codyco-superbuild/build/install -C/Users/jorhabibeljaik/Code/codyco-superbuild/build/external/urdfdom/CMakeFiles/YCMTmp/urdfdom-cache.cmake -GXcode /Users/jorhabibeljaik/Code/codyco-superbuild/external/urdfdom
Not searching for unused variables given on the command line.
loading initial cache file /Users/jorhabibeljaik/Code/codyco-superbuild/build/external/urdfdom/CMakeFiles/YCMTmp/urdfdom-cache.cmake
-- The CXX compiler identification is AppleClang 7.3.0.7030031
-- The C compiler identification is AppleClang 7.3.0.7030031
-- Check for working CXX compiler using: Xcode
-- Check for working CXX compiler using: Xcode -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Xcode
-- Check for working C compiler using: Xcode -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- urdfdom version 0.4.0
-- Performing Test HAS_VISIBILITY
-- Performing Test HAS_VISIBILITY - Success
-- Found TinyXML: /usr/local/lib/libtinyxml.dylib  
CMake Error at CMakeLists.txt:41 (find_package):
  Could not find a configuration file for package "urdfdom_headers" that is
  compatible with requested version "0.4".

  The following configuration files were considered but not accepted:

    /Users/jorhabibeljaik/Code/codyco-superbuild/build/install/share/urdfdom_headers/cmake/urdfdom_headers-config.cmake, version: unknown

-- Configuring incomplete, errors occurred!
See also "/Users/jorhabibeljaik/Code/codyco-superbuild/build/external/urdfdom/CMakeFiles/CMakeOutput.log".
make: *** [/Users/jorhabibeljaik/Code/codyco-superbuild/build/external/urdfdom/CMakeFiles/YCMStamp/Debug/urdfdom-configure] Error 1
Command /bin/sh failed with exit code 2

** BUILD FAILED **
```

I have checked the version stated in the `CMakeLists` of `urdfdom` and while it has the required version 0.4.0, `urdfdom-headers` does not as you can see next, but 0.3.0.

```bash
cmake_minimum_required( VERSION 2.8 FATAL_ERROR )
project (urdfdom_headers)

set (URDF_MAJOR_VERSION 0)
set (URDF_MINOR_VERSION 3)
set (URDF_PATCH_VERSION 0)

set (URDF_VERSION ${URDF_MAJOR_VERSION}.${URDF_MINOR_VERSION}.${URDF_PATCH_VERSION})

message (STATUS "${PROJECT_NAME} version ${URDF_VERSION}")
[...]
```

I have done `git pull` in `external/urdfdom_headers` but it remains the same.

Here's the output of `git log -n 1`

```bash
commit 8508d415638af4099b22ad54ffd36e0b0bf4e575
Author: Silvio Traversaro <silvio.traversaro@iit.it>
Date:   Sun Oct 4 22:33:59 2015 +0200

    Fix handling rpy with pitch equal to +/-M_PI/2
```

traversaro, 2016-05-10:
Mhhh... it is possible that you have a `urdfdom` installed from homebrew  ? I think that in this case you can tell the superbuild to download its own urdfdom/urdfdom_headers by setting the option USE_SYSTEM_URDFDOM / USE_SYSTEM_URDFDOM_HEADERS to OFF in the the CMake of the superbuild.

jeljaik, 2016-05-10:
**Solution**

Apparently I had  a version different from the one the superbuild should have downloaded, which required `urdfdom_headers 0.4`. By deleting `urdfdom` from `external` and building the superbuild again, the right one was downloaded and everything went as expected. 

Thanks

closed 2016-05-10 by jeljaik