robotology-legacy/codyco-superbuild #130: WebHooks for Ocra

ahoarau, 2016-05-21:
Would it be possible to trigger a superbuild whenever we commit to ocra's master branch ?

traversaro, 2016-05-22:
That would be a great idea, I investigated it in the past but I did not found an easy way to do it (but it is possible that it is quite easy, just I did not investigated enough). In the meanwhile I asked if you can get access to the codyco-superbuild , so you could relaunch the `codyco-superbuild` Travis build manually as many times as you want.

traversaro, 2016-05-22:
P.s. at the moment we are launching the Travis build once a day thanks to https://nightli.es/ . 
By the way I just noticed that now Travis support cron jobs ( https://github.com/travis-ci/travis-ci/issues/582#issuecomment-218095192 ), so we may want to try them sooner or later.

traversaro, 2016-05-27:
Related Travis upstream issue : https://github.com/travis-ci/travis-ci/issues/249 .

ahoarau, 2016-05-27:
I'm fine with the nightli :=) , so I'm closing this. Thanks for the clarification !

closed 2016-05-27 by ahoarau