robotology-legacy/codyco-superbuild #135: Error since the update of codyco-superbuild

misaki43, 2016-06-02:
Hello,

I am not sure it is linked to my update of codyco-superbuild but I have some issues:
1. When I do the "ccmake" in the build folder, the path of the directories used are not found. It is strange because I haven't modified the "bashrc_icub".
2. When I do the "make", I have the error: 
`CMake Error: The source directory "/home/odermy/Software/codyco-superbuild/main/icub-gazebo" does not appear to contain CMakeLists.txt.`
3. Before that,  KinCartesianSolver wasn't found.

Could you help me please?

traversaro, 2016-06-02:
1. You mean that the codyco-superbuild is not finding the YARP and ICUB libraries pointd by `YARP_DIR` and `ICUB_DIR`? Are you sure that this enviromental variables are properly defined?

2. Did you enabled the `CODYCO_USES_GAZEBO` flag? Please disable it as it is still an experimental option. After you disabled it please delete the `main/icub-gazebo` and the `build/main/icub-gazebo` directories to solve the problem.

3. iKinCartesianSolver is compiled by `icub-main`, so check where you compiled `icub-main`, if using the super build or externally as you mentioned before.

misaki43, 2016-06-03:
1. Not only them, all the libraries are not found (WBIToolboxControlers_Dir, etc), I don't know if it is normal.
For the environment I don't change the bashrc:

> export CODYCO_SUPERBUILD_ROOT=~/Software/codyco-superbuild
export PATH=$PATH:$CODYCO_SUPERBUILD_ROOT/build/install/bin
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$CODYCO_SUPERBUILD_ROOT/build/install/lib
export YARP_DATA_DIRS=$YARP_DATA_DIRS:$CODYCO_SUPERBUILD_ROOT/build/install/share/codyco:$CODYCO_SUPERBUILD_ROOT/build/install/share/yarp:$CODYCO_SUPERBUILD_ROOT/build/install/share/iCub:$CODYCO_SUPERBUILD_ROOT/build/install/share/WB-Toolbox

2. Youps, it was the problem, sorry and thanks!

3. Normally it is done with the superbuild. I don't change the installation since Darmstadt, I just tried to update the project (and since I had problem I change the ccmake and add  foolishly the "CODYCO_USES_GAZEBO flag").
Now, when I do the make of the codyco superbuild i have this error: 

> loading initial cache file /home/odermy/Software/codyco-superbuild/build/external/ICUB/CMakeFiles/YCMTmp/ICUB-cache.cmake
> loading initial cache file /home/odermy/Software/codyco-superbuild/build/libraries/paramHelp/CMakeFiles/YCMTmp/paramHelp-cache.cmake
> CMake Error at app/robots/CMakeLists.txt:56 (add_subdirectory):
>   add_subdirectory given source "icubGazeboSim" which is not an existing
>   directory.
> CMake Warning at app/robots/CMakeLists.txt:77 (message):
>   ROBOT_NAME is set to icubGazeboSim but the app/robots/icubGazeboSim
>   directory does not exist

when I look in external/ICUB/app/robots this robot is not here. I try with the "iCubDarmstadt01" just to see if it works better, but I have the same result 

>  CMake Warning at app/robots/CMakeLists.txt:77 (message):
   ROBOT_NAME is set to icubDarmstadt01 but the app/robots/icubDarmstadt01
   directory does not exist

misaki43, 2016-06-06:
1. In fact I have a question: since YARP and ICUB are installed with codyco-superbuild, should I precise something in the bashrc?

traversaro, 2016-06-06:
Sorry @misaki43 , I was away for the weekend and my plane got canceled. : \
Going back to the questions: 
* The fact that during `ccmake` you see `iDynTree_DIR-NOTFOUND` in the cmake windows means that those libraries were not found in the system by the superbuild, and hence they will be downloaded and installed by the superbuild itself. To help clarify this, you should find a message like this if you type `cmake .` in the superbuild build directory, that explains which project have  been found (and so they will not be downloaded and installed by the superbuild) and which project have not been found (and so they will be downloaded, compiled and installed by the superbuild).
* To have the YARP and ICUB binaries installed by the superbuild in your PATH, you should modify your bashrc as discussed in https://github.com/robotology/codyco-superbuild/#configure-your-environment . 
* The `ROBOT_NAME` related error that you see should be something that have been recently solved in `icub-main` (in https://github.com/robotology/icub-main/pull/322). Update the superbuild (following the instructions in https://github.com/robotology/codyco-superbuild#update) and try again.

misaki43, 2016-06-06:
Hi, no problem and thanks for your help! 
* Yes correct, so the "not found library" is not a problem.
* So my path seems to be correct.
* Ok exact, thanks!

All of these problems appeared when I tried to update the codyco-superbuild before compiling some part of the project. But in fact, there was only one real error (ROBOT_NAME) and some problem of comprehension. 

I have two other issues (that I tried to solve before all of that). I didn't create a new issue because maybe these are simple misunderstandings:
1.  On the [cartesian interface](http://wiki.icub.org/iCub/main/dox/html/icub_cartesian_interface.html) tutorial, you can read: 
> On the cluster: compile the repository with the switch ENABLE_icubmod_cartesiancontrollerclient 

But on which repository can I do that?  (Since Icub-main is installed with the codyco-superbuild, how can I change the flags?)
2. I also need to activate the ICUB_USE_IPOPT flag, but I assume it is the "same problem"
(We can also chat on skype to solve those problems directly if you prefer)

traversaro, 2016-06-07:
To change the flags of the `icub-main` build by the codyco superbuild, just go (i.e. `cd`) in the `build/external/ICUB` and launch `ccmake` from there. Then you can modify all the flags as if you were in a normal build directory. However those flags should be already setted by the superbuild when configuring `icub-main`, see https://github.com/robotology/codyco-superbuild/blob/master/cmake/BuildICUB.cmake#L13 .

closed 2016-06-08 by traversaro