robotology-legacy/codyco-superbuild #137: build with catkin

ahoarau, 2016-06-07:
I'm experiencing building the superbuild with catkin tools in order to have a ROS workspace containing all the icub/ocra related stuff.

I can add a simple package.xml : 
``` xml
<?xml version="1.0"?>
<package>
  <name>codyco-superbuild</name>
  <version>1.0.0</version>
  <description>The Codyco Superbuild package</description>
  <maintainer email="iit@iit.it">iit people</maintainer>
  <license>GPLv2</license>

  <buildtool_depend>cmake</buildtool_depend>

</package>
```
 And it build **fines** !

Is there any option to put all the generated libraries in `codyco_ws/devel/codyco-superbuild/` instead of `codyco_ws/build/codyco-superbuild/install` ?

traversaro, 2016-06-07:
I briefly tried to find how to do it, but I was not able to find a way. Probably @drdanz can say if it is possible or not (I guess https://github.com/robotology/ycm/issues/18 is related, but I am not sure). 

Another possible strategy for using `catkin` (or in the future `ament`) is to actually provide `package.xml` for all the subprojects, and make it possible to build them with both catkin and the traditional CMake superbuild. Something in that direction (I guess) was discussed in https://github.com/robotology/robotology.github.io/issues/1 , but again @drdanz can discuss this more in detail.

drdanz, 2016-06-07:
I have no idea and I'm no expert with `catkin`/`ament`...
Have you already tried adding `CMAKE_ARGS -DCMAKE_INSTALL_PREFIX=<something>` In the `Build<Package>.cmake` file?
Can I see the relative `Build<Package>.cmake` file?

traversaro, 2016-06-07:
@drdanz the question is actually unrelated to `catkin`/`ament`: it is possible to install the stuff in the superbuild to another location that is not `SUPERBUILD_ROOT/build/install`?

drdanz, 2016-06-07:
If you mean "all the stuff in the superbuild", not at the moment, sorry... It would not be difficult to do it, but there are a few issues with allowing to install into a different install prefix.
Basically, the installation outside the build directory should happen only when you run "make install", since the install prefix could be writable only by root, and it would be a problem to have to run "sudo make"

Actually I just had an idea of how to implement it, but I don't think I will have the time to implement it for quite a while.

ahoarau, 2016-06-07:
@drdanz that's fine, I don't really need it right now, but that would simplify some builds in the future

traversaro, 2016-06-07:
Ok, I think that then we can close this issue because it would be a duplicate of https://github.com/robotology/ycm/issues/18 .

closed 2016-06-07 by traversaro