robotology-legacy/codyco-superbuild #138: cmake cannot find already installed ICUB_DIR and gazeboYarpPlugin

serena-ivaldi, 2016-06-10:
Hello

I am installing a brand new machine step by step, detailing progress here: https://github.com/inria-larsen/icub-manual/wiki/How-to-install-the-software-on-your-machine
I have installed yarp, icub-main, gazeb-yarp-plugins etc and everything works fine, iCub in Gazebo works.
I am now installing the codyco-superbuild but when doing ccmake many projects are not found, in particular it does not find ICUB_DIR while YARP is found:

ICUB_DIR                        *ICUB_DIR-NOTFOUND                              
YARP_DIR                        */home/icub/software/src/yarp/build             

and when I set CODYCO_USES_GAZEBO on, then also gazebo plugins are not found:

GazeboYARPPlugins_DIR            GazeboYARPPlugins_DIR-NOTFOUND

I tried entering manually their paths in the ccmake but it does not work. I don't want to compile now with the risk that it will download and compile another version of these modules...

How can I fix this issue?
Thanks in advance.

traversaro, 2016-06-10:
Hi serena: 
 * the `CODYCO_USES_GAZEBO` is still an experimental option, and for this reason is not mentioned in the README, so please don't use it until it is fully tested. 
 * For `ICUB_DIR` : this is strange. Have you tried to compile the `icub-tutorials` ( https://github.com/robotology/icub-tutorials ) with the same configuration you use to compile the superbuild?

serena-ivaldi, 2016-06-10:
Hi Silvio,

ok for the experimental flag :)

For the tutorials, we already compiled some of them for testing and they work. But it was individual compilation, like going into motorControlAdvanced, creating a build, ccmake...
Do you mean to compile all of them using the CmakeList inside /src?

serena-ivaldi, 2016-06-10:
I did it by entering icub-tutorials, creating a build and sourcing the CMakeList... it successfully finds ICUB_DIR

traversaro, 2016-06-10:
Mhhh.. I think I got it. Can you try to update `icub-main`? The codyco-superbuild requires icub at least to be of version `1.4.0`, and I think that not finding it (because it finds an earlier version) it decides that it need to download it. 
See this line : https://github.com/robotology/codyco-superbuild/blob/master/CMakeLists.txt#L97 .

serena-ivaldi, 2016-06-10:
:,(

I tried.
I just git pulled icub-main and recompiled and installed, and recompiled codyco-superbuild and stilll it does not find ICUB_DIR.

traversaro, 2016-06-10:
Apparently there was a problem, because ICUB does not exports its version in the CMake configuration files and thus the check for the version always fails. 
This commit disable the check : https://github.com/robotology/codyco-superbuild/commit/1cc9e25089cba094ccc33d2e5297b0dba1c8ecee and so if you update the superbuild now everything should work.

serena-ivaldi, 2016-06-13:
thanks for the fix, it found it !

closed 2016-06-13 by serena-ivaldi