robotology-legacy/codyco-superbuild #141: Matlab is not able to load required libraries

gabrielenava, 2016-07-11:
For some versions of Ubuntu and Matlab, the toolbox which requires Matlab such as WBToolbox, WBIToolboxControllers and IDynTree complain about missing libraries. Here is the output for WBToolbox:

```
Invalid MEX-file '/home/DanielePucci/Software/codyco-superbuild/build/install/mex/WBToolbox.mexa64':
 /home/DanielePucci/Software/matlab/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6:
version `GLIBCXX_3.4.21' not found (required by /home/DanielePucci/Software/codyco-superbuild/build/install/mex/WBToolbox.mexa64)
```
This issue has been temporarily solved adding to the .bashrc 

```
alias matlab="LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 matlab"
```
The error comes out with Ubuntu xenial, Matlab version is 2016a and gcc version  is 5.3.1 20160413

traversaro, 2016-08-03:
I solved the problem in a more permanent way my renaming the matlab's `libstdc++.so.6` to a different name:
~~~
sudo mv  /usr/local/MATLAB/R2016a/sys/os/glnxa64/libstdc++.so.6 /usr/local/MATLAB/R2016a/sys/os/glnxa64/libstdc++.so.6.back
~~~
In this way matlab always finds the system `libstdc++.so.6` instead of its own version, regardless of how you call it. 
This is extremely important to make sure that `matlab` works correctly even when is not called by the terminal, for example when it is called by `CTest` while running some automatic tests for some repositories.

francesco-romano, 2016-11-02:
Every now and then these runtime issues pop up.

An additional alternative, which probably can be the most clean of the two previous, is to copy the file in `matlabroot/bin/.matlab7rc.sh` into the user's home folder, e.g. on macOS
`cp /Applications/MATLAB_R2016b.app/bin/.matlab7rc.sh ~/.matlab7rc.sh` and modifying it by looking for the corresponding architecture section (e.g. glnx* for linux, ...) and add the libraries to the `LDPATH_PREFIX` variable. 
For example in my case:
`LDPATH_PREFIX='/usr/local/opt/gcc/lib/gcc/6/libgfortran.3.dylib'`

diegoferigo, 2017-07-06:
This workaround is still required

Ubuntu Xenial
Matlab 2016b
GCC 5.4.0 20160609

traversaro, 2017-07-21:
cc @fjandrad

traversaro, 2017-08-10:
This information need to be moved to some FAQ or QA issue.

traversaro, 2018-04-19:
Moved the problem of properly documenting this at the superbuild level in https://github.com/robotology/robotology-superbuild/issues/64 .

closed 2018-04-19 by traversaro

Yeshasvitvs, 2018-06-06:
I tried to run the standup controller from whole-body-controllers on a new installation of MATLAB Version 9.4 (R2018a), Simulink Version 9.1 (R2018a). While trying to run the simulink model "unable to run the MATLABWindow application" warning message with other error messages are thrown and the model was terminated without running. @gabrielenava and I followed this [issue](https://it.mathworks.com/matlabcentral/answers/364551-why-is-matlab-unable-to-run-the-matlabwindow-application-on-linux) and renamed all the `libglib-*` files to `libglib-*.old`. This solved the issue and the simulink model is running without errors now.

traversaro, 2018-11-14:
@Yeshasvitvs on which version of Ubuntu you had this problem?

Yeshasvitvs, 2018-11-14:
@traversaro Ubuntu 18.04