robotology-legacy/codyco-superbuild #142: CMake configure failing with ccmake

TeoKa, 2016-07-12:
~~~
 CMake Error at /usr/local/Cellar/cmake/3.5.2/share/cmake/Modules/FeatureSummary.cmake:304
 (CMAKE_PARSE_ARGUMENTS):
   Unknown CMake command "CMAKE_PARSE_ARGUMENTS".
 Call Stack (most recent call first):
   cmake/YCMBootstrap.cmake:76 (set_package_properties)
   CMakeLists.txt:84 (include)
~~~

jeljaik, 2016-08-25:
I have the exact same error today after updating `codyco-superbuild` on my OS X setup. Any hint guys? 

`cmake version 3.5.2`

traversaro, 2016-08-25:
Run : 
~~~
cmake .
~~~
without passing throught ccmake is a workaround for this issue.

jeljaik, 2016-08-25:
Yeah sure that's what I did. Still thought it would be cool to let you guys
know. A bit uncomfortable when you're looking for some specific options to
pass.

traversaro, 2016-08-25:
Ah, got it! Anyhow writing down the workaround may be useful for other people. For the issue itself we are unable to reproduce it on a simpler model at the moment.

traversaro, 2016-09-09:
We got the same problem on Windows :'''''( .
@MartaLorenzini can you paste the screenshot here?

traversaro, 2016-09-09:
In our case, restarting `cmake-gui` solved the problem.

jeljaik, 2016-09-20:
Could it be related to what's mentioned [here](https://github.com/dmlc/mxnet/issues/1131) or [here](https://cmake.org/Bug/bug_relationship_graph.php?bug_id=14230&graph=dependency)?  a possibly missing `include(CMakeParseArguments)` in some YCM `cmake` file?

traversaro, 2016-09-20:
I don't think that is the case, as from CMake 3.5 `cmake_parse_arguments` is a built-in command and `CMakeParseArguments` is just an empty placeholder (see https://github.com/Kitware/CMake/blob/v3.5.0/Modules/CMakeParseArguments.cmake ).

traversaro, 2018-04-18:
I did not heard about anyone experiencing this problem in the last ~1.5 years, I will close this, please reopen (perhaps in https://github.com/robotology/robotology-superbuild) if necessary.

closed 2018-04-18 by traversaro