robotology-legacy/codyco-superbuild #150: CMake errors with version 2.8.11 -> 2.8.12 Debian 7

rlober, 2016-08-24:
We were trying to compile superbuild on our cluster blades (Debian 7) and ran into the following error during the cmake phase of iDynTree:
```
[ 31%] Performing configure step for 'iDynTree'
Not searching for unused variables given on the command line.
loading initial cache file /usr/local/src/robot/source/codyco-superbuild/build-x86_64/libraries/iDynTree/CMakeFiles/YCMTmp/iDynTree-cache.cmake
-- Using iCub from build tree
-- Boost version: 1.49.0
CMake Error at src/estimation/CMakeLists.txt:33 (target_compile_options):
  Unknown CMake command "target_compile_options".

-- Configuring incomplete, errors occurred!
make[2]: *** [libraries/iDynTree/CMakeFiles/YCMStamp/iDynTree-configure] Error 1
make[1]: *** [CMakeFiles/iDynTree.dir/all] Error 2
make: *** [all] Error 2
```

The apt version of cmake for wheezy is 2.8.11 and `target_compile_options` was introduced in 2.8.12: http://cmake.blogspot.fr/2013/10/cmake-2812-released.html

After upgrading to cmake 2.8.12 from source we now get the following issue:

```
-- The C compiler identification is GNU 4.7.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- YCM not found. Bootstrapping it.
-- Downloading YCMEPHelper.cmake - Attempt 1 of 3
-- Downloading YCMEPHelper.cmake - ERROR 1: "unsupported protocol"
-- Downloading YCMEPHelper.cmake - Attempt 2 of 3
-- Downloading YCMEPHelper.cmake - ERROR 1: "unsupported protocol"
-- Downloading YCMEPHelper.cmake - Attempt 3 of 3
CMake Error at cmake/IncludeUrl.cmake:352 (message):
  Downloading YCMEPHelper.cmake - ERROR 1: "unsupported protocol"
Call Stack (most recent call first):
  cmake/YCMBootstrap.cmake:105 (include_url)
  CMakeLists.txt:84 (include)

-- Configuring incomplete, errors occurred!
See also "/usr/local/src/robot/source/codyco-superbuild/build-x86_64/CMakeFiles/CMakeOutput.log".
```

I know this is more an iDynTree issue, but seeing as how it kills the default superbuild build I think it should just be changed at the superbuild level.

traversaro, 2016-08-24:
For the first one, it should have been fixed by https://github.com/robotology/idyntree/commit/e75a3f7887cb4c75fc3a54a98935e90f57405146 (you can update the superbuild and try). 
Unfortunately the CI is not covering CMake 2.8.11 at the moment (see https://github.com/robotology/idyntree/issues/111), that's why this kind of problems are not properly identified before merging.

The second one is actually a problem of CMake compiled without SSL support (that is the default when you compile CMake from source), see https://github.com/robotology/codyco-superbuild/issues/26#issuecomment-231011954 . Clearly the upstream error could be more descriptive.

jeljaik, 2016-08-24:
Merci! 😉 That worked for us.

closed 2016-08-24 by traversaro

rlober, 2016-08-24:
Bellissimo pasta pizza!

rlober, 2016-08-24:
:smile: Thanks @traversaro