robotology-legacy/codyco-superbuild #152: cannot start wholeBodyDynamicsTree because of inertial error

serena-ivaldi, 2016-09-30:
Dear all

we are trying to run wholeBodyDynamicsTree for Gazebo and the program does not start on a weird error:
wholeBodyDynamicsTree --autoconnect --headV2

```
yarp: Port /wholeBodyDynamicsTree/ftSens/r_foot_ft_sensor:i active at tcp://10.0.0.11:10072
yarp: Receiving input from /icubGazeboSim/right_foot/analog:o to /wholeBodyDynamicsTree/ftSens/r_foot_ft_sensor:i using udp
yarp: Port /wholeBodyDynamicsTree/imu/imu_frame:i active at tcp://10.0.0.11:10073
yarpWholeBodySensors::openImu():  could not connect /icubGazeboSim/inertial to /wholeBodyDynamicsTree/imu/imu_frame:i
[ERR] yarpWholeBodySensors::init() error: failing in opening imu sensors.
[ERROR]wholeBodyDynamicsTree : Error while initializing whole body estimator interface.Closing module 
```

We just pulled and updated yarp, icub-main, codyco-superbuild, the gazebo plugin, even gazebo7.
The YARP_ROBOT_NAME is icubGazeboSim.

We also cannot access to the port /inertial in gazebo apparently:
```
icub@icub01:~$ yarp read ... /icubGazeboSim/inertial
                                                                                                                                          
│yarp: Port /tmp/port/2 active at tcp://10.0.0.11:10078
                                                                                                                                          
│Failure: Do not know how to connect to tcp://tmp/port/2

```

How can we fix this?
Thanks in advance

@paraschos @misaki43

traversaro, 2016-09-30:
`icub@icub01:~$ yarp read ... /icubGazeboSim/inertial` is working fine for me. 

From the error `|Failure: Do not know how to connect to tcp://tmp/port/2` I strongly suspect you have problem related to the YARP network setup. 

On which machine are you running the `yarpserver`? How many network interfaces there are in the PC on Gazebo is running?

serena-ivaldi, 2016-09-30:
only one interface, and it is on the same machine.. 
we can read all the other ports except the inertial

traversaro, 2016-09-30:
Just to be sure, can you start the `yarpserver`, launch gazebo and try to read the `/icubGazeboSim/inertial` port and then copy paste in this issue the `yarpserver` output?

misaki43, 2016-10-03:
Hello,
today, when I launched gazebo, it worked... The only difference since the issue is that the computer has been power off/on, maybe something has been initialized at that moment (?).
Sorry for that issue!

traversaro, 2016-10-03:
Perfect, closing the issue then.

closed 2016-10-03 by traversaro