robotology-legacy/codyco-superbuild #155: Document robot-specific files necessary for CoDyCo software

traversaro, 2016-10-03:
See https://github.com/robotology/icub-support/issues/304 . 

At the moment I think that each robot needs (at the moment): 
* A `.ini` and `.urdf` model in [`yarp-wholebodyinterface`](https://github.com/robotology/yarp-wholebodyinterface/tree/master/app/robots) ,
* A `.ini` configuration file for [`torqueBalancing`](https://github.com/robotology/codyco-modules/tree/fb9952099bea4f7be55995edb691fb47b457de87/src/modules/torqueBalancing/app/robots) ,
* A `.ini` configuration file for [`wholeBodyDynamicsTree`](https://github.com/robotology/codyco-modules/tree/fb9952099bea4f7be55995edb691fb47b457de87/src/modules/wholeBodyDynamicsTree/app) ,
* A `.m` gain configuration file for each controller in [`WBI-Toolbox-controllers`](https://github.com/robotology-playground/WBI-Toolbox-controllers/tree/master/controllers) .

Note that most of this files could be removed if we had a way to just refer to some "template" file, that is ok for each "class" of robot (for example all the iCub 2.5 Eth robot at the moment share almost identical conf files). In fact most of the configuration have been created by copy-pasting the conf file for the most similar robot available at the moment, and this is error-prone (see https://github.com/robotology/codyco-modules/issues/202). A good tradeoff would be to provide for some classes of robots (for example for iCub 2.5 ETH, i.e. any new robot) a template that can be initially copied for any new robot. 

cc 
@serena-ivaldi @olivierrochel-inria @dogoepp

traversaro, 2016-10-03:
Another problems is that the performance given by a gain configuration file is typically (unfortunately) high dependent on the underlying torque control parameters, that are themselves highly dependent on the motors mounted on the specific iCub, that unfortunately can be different even in robots of the same "class" (i.e. two different iCub 2.5 Eth). For this reason the only actually reliable way to get gain parameters ensuring a given level of performance is by tuning on the actual robot.

misaki43, 2016-10-04:
Ok but it is not so simple: 
For example, about the WholeBodyInterface stuff, we need to add all files of a new folder "/app/robots/icubNancy01" and change the CMakeFiles by adding our robot in.
It seems to be exactly the same for torqueBalancing.
If you see other things we have to change, please tell us, it will help us a lot to change all files directly.
 
About the gain file, I remember all the tunings that have been done in Darmstadt, that seemed to be complicated and we guess that our lack of knowledge about it could create errors and could damage the robot. So we are a little worried to do it: is it possible to have a complete tutorial about that, or directly someone to come here to take care of that tuning? (in that case, if someone patient come to do it, I could create a complete tutorial about it :))

traversaro, 2018-04-19:
Fortunately, the amount of necessary configuration files is reducing over time. : ) 
I think all the relevant information can be integrated in the document mentioned here: https://github.com/robotology/codyco-superbuild/issues/118 .

closed 2018-04-19 by traversaro