robotology-legacy/codyco-superbuild #160: Postmortem of CoDyCo-related YOGA++ failure on 2016/12/16

traversaro, 2016-12-16:
See https://en.wikipedia.org/wiki/Postmortem_documentation.

Attached the log of run of yarprobotinterface : 

[yarprunlog_16_12_2016_12_38_17.log.zip](https://github.com/robotology/codyco-superbuild/files/657093/yarprunlog_16_12_2016_12_38_17.log.zip)

traversaro, 2016-12-16:
cc @gabrielenava @DanielePucci

traversaro, 2016-12-16:
Relevant part of the log. The problem seems to start around 8641 s and at 8642,46 the controller has been stopped.

~~~
26781	8628,155456	INFO	icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
26782	8628,165596	INFO	icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
26783	8629,300335	DEBUG	cfw2can [0] thread ran 595 times, req.dT:10[ms], av.dT:10.08[ms] av.loopT :0.02[ms]
26784	8629,300405	DEBUG	     Device has no ICanBusErr interface
26785	8629,300430	DEBUG	cfw2can [0] printing boards infos:
26786	8629,300450		None
26788	8633,204785	INFO	icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
26789	8633,214939	INFO	icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
26791	8635,305973	DEBUG	cfw2can [0] thread ran 594 times, req.dT:10[ms], av.dT:10.09[ms] av.loopT :0.03[ms]
26792	8635,306106	DEBUG	     Device has no ICanBusErr interface
26793	8635,306151	DEBUG	cfw2can [0] printing boards infos:
26794	8635,306189		None
26795	8636,311220	ERROR	 from BOARD 10.0.1.6 (EB6), src CAN2, adr 13, time 1265s 135m 69u: (code 0x01000002, par16 0x0003 par64 0x000000000000006d) -> HW: strain values saturation. In p16 there is the channel involved, in p64 there are the saturation counts for the last sec of both types (lower: LS32bits, upper MS32bits + .
26796	8636,311347	ERROR	 from BOARD 10.0.1.6 (EB6), src CAN2, adr 13, time 1265s 135m 199u: (code 0x01000002, par16 0x0004 par64 0x0000004200000000) -> HW: strain values saturation. In p16 there is the channel involved, in p64 there are the saturation counts for the last sec of both types (lower: LS32bits, upper MS32bits + .
26798	8637,385567	ERROR	 from BOARD 10.0.1.8 (EB8), src CAN2, adr 13, time 1266s 222m 885u: (code 0x01000002, par16 0x0003 par64 0x0000000000000058) -> HW: strain values saturation. In p16 there is the channel involved, in p64 there are the saturation counts for the last sec of both types (lower: LS32bits, upper MS32bits + .
26799	8637,385649	ERROR	 from BOARD 10.0.1.8 (EB8), src CAN2, adr 13, time 1266s 223m 18u: (code 0x01000002, par16 0x0004 par64 0x000000da00000000) -> HW: strain values saturation. In p16 there is the channel involved, in p64 there are the saturation counts for the last sec of both types (lower: LS32bits, upper MS32bits + .
26800	8638,246071	INFO	icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
26801	8638,256174	INFO	icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
26802	8639,542057	ERROR	 from BOARD 10.0.1.9 (EB9), src CAN2, adr 1, time 1268s 374m 880u: (code 0x01000002, par16 0x0000 par64 0x0000000000000005) -> HW: strain values saturation. In p16 there is the channel involved, in p64 there are the saturation counts for the last sec of both types (lower: LS32bits, upper MS32bits + .
26804	8641,304045	DEBUG	cfw2can [0] thread ran 594 times, req.dT:10[ms], av.dT:10.09[ms] av.loopT :0.03[ms]
26805	8641,304105	DEBUG	     Device has no ICanBusErr interface
26806	8641,304123	DEBUG	cfw2can [0] printing boards infos:
26807	8641,304137		None
26808	8641,466068	ERROR	 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 1270s 293m 497u: (code 0x02000000, par16 0x0000 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .
26809	8641,466138	ERROR	 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 1270s 293m 610u: (code 0x02000000, par16 0x0002 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .
26810	8641,466160	ERROR	 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 1270s 293m 709u: (code 0x02000000, par16 0x0003 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .
26811	8641,466180	DEBUG	cfw2can [0] msg from board 1: PWM DIS CH1
26812	8641,466198	DEBUG	cfw2can [0] msg from board 1: PWM DIS CH0
26813	8641,476148	DEBUG	cfw2can [0] msg from board 2: FAULT AXIS: 0 & 1 -16384
26814	8641,476209	ERROR	 from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 1270s 305m 314u: (code 0x02000000, par16 0x0000 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .
26815	8641,476232	ERROR	 from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 1270s 305m 426u: (code 0x02000000, par16 0x0002 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .
26816	8641,476248	ERROR	 from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 1270s 305m 525u: (code 0x02000000, par16 0x0003 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .
26817	8641,476266	ERROR	 from BOARD 10.0.1.5 (EB5), src LOCAL, adr 0, time 1270s 301m 179u: (code 0x02000000, par16 0x0000 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .
26818	8641,476280	ERROR	 from BOARD 10.0.1.5 (EB5), src LOCAL, adr 0, time 1270s 301m 272u: (code 0x02000000, par16 0x0002 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .
26819	8641,476295	ERROR	 from BOARD 10.0.1.7 (EB7), src LOCAL, adr 0, time 1270s 295m 241u: (code 0x02000000, par16 0x0000 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .
26820	8641,476309	ERROR	 from BOARD 10.0.1.7 (EB7), src LOCAL, adr 0, time 1270s 295m 336u: (code 0x02000000, par16 0x0001 par64 0x0000000000000000) -> MC: exernal fault button pressed. + .
[... cut ...]
26946	8666,391075	ERROR	 from BOARD 10.0.1.6 (EB6), src CAN2, adr 13, time 1295s 212m 158u: (code 0x01000002, par16 0x0004 par64 0x000003e900000000) -> HW: strain values saturation. In p16 there is the channel involved, in p64 there are the saturation counts for the last sec of both types (lower: LS32bits, upper MS32bits + .
26948	8668,505953	INFO	icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
26949	8668,516011	INFO	icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
~~~

gabrielenava, 2016-12-16:
Description of the failure: the robot was performing the Yoga ++ demo. After switching from left foot to right foot balancing, while the robot was doing yoga movements with the arms, tippo-tappo started. I pressed the "stop" button in Simulink, but apparently the robot went in hardware fault. We suspect this happened because of fault cable's jack is broken. This also explains why, according to what @DanielePucci said, the robot joints were in idle after the failure. Furthermore, one of the leg's joint was in hardware fault.

traversaro, 2016-12-16:
Thanks @gabrielenava , I think this is the most advanced postmortem we can achieved given the data that we have.

closed 2016-12-16 by traversaro