robotology-legacy/codyco-superbuild #166: [wholeBodyDynamics] Provide force-torque offset calibration methods that take as input the calibration frames

traversaro, 2017-02-16:
Note that for calibration assuming that only an external force-torque is acting on the robot, it would be sufficient to specify the **link** on which the external force-torque is exerted, while if we are performing the calibration assuming that two external force-torques are acting on the robot, we need to specify the frames w.r.t. the external force-torque least squares estimation problem is solved. 

cc @gabrielenava @DanielePucci

DanielePucci, 2017-02-21:
Nice feature. Would it be possible to make this feature number-of-frame independent? This may be useful for calibration during the latest development of the balancing demos, i.e. robot sitting on the chair and feet on ground, which has more than two links in contact with the environment.

traversaro, 2017-02-21:
The first implementation is available at https://github.com/robotology/codyco-modules/pull/221 . 

I don't think it make sense to implement it for an arbitrary number of links in contact, before verying  that it "works" under some sort of conditions (and perhaps changing the algorithm to estimate the external force-torque, as discussed f2f friday at the whiteboard).

DanielePucci, 2017-02-22:
Roger. Let's see if this new feature works, and then we'll see if it worth to extend it.

traversaro, 2018-04-18:
The feature seem to be working fine, if users (@fiorisi @nunoguedelha @fjandrad @S-Dafarra @gabrielenava ) are interested in more development on this side please open an issue in the corresponding repo, thanks!

closed 2018-04-18 by traversaro