robotology-legacy/codyco-superbuild #167: Travis builds on macOS started failing because the time quota is over

traversaro, 2017-03-09:
Something changed (perhaps in homebrew, a formula was converted from a bottle to a source formula) and now the build take much more time.

traversaro, 2017-03-09:
cc @diegoferigo that it getting a lot of insight on this.

diegoferigo, 2017-03-10:
It might be, the deps installing step takes much more time than the previous builds. The problem here is that Travis on macOS images has a 50 minutes limit per job (on linux the limit is 120 minutes). Building all the components from sources, including the ones outside the codyco-specific pieces of sw, and being already on the edge, may let the time approach that limit as in this case. Even not having any kind of experience with homebrew, not having any control on what is shipped as bottle I think is a constraint we cannot overcome.

I see two possible solutions:
* `ccache` looks enabled for the codyco-superbuild objects. Question: is it used also when homebrew builds the formulas? If not, we could exploit it
* Handle the [homebrew cache](https://github.com/Homebrew/brew/blob/master/docs/FAQ.md#where-does-stuff-get-downloaded) with Travis persistent cache. I'm not sure if it can cut time only when downloading the sources or also when building the formulas

@francesco-romano Did you try to create a `yarp` bottle with the attempt you mentioned last week? Do I remember well?

traversaro, 2017-03-15:
cc @francesco-romano

closed 2017-03-15 by traversaro