robotology-legacy/codyco-superbuild #172: Travis failure on Linux

traversaro, 2017-05-09:
See https://travis-ci.org/robotology/codyco-superbuild/builds/230186141 .

traversaro, 2017-05-09:
The problem is in `ocra-wbi-plugins` now depends on GUROBI: 
~~~
CMake Error at /home/travis/build/robotology/codyco-superbuild/build/install/share/YCM/cmake-next/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find GUROBI (missing: GUROBI_LIBRARY GUROBI_CXX_LIBRARY
  GUROBI_INCLUDE_DIR)
~~~

@rlober @jeljaik Given the problems related to installing GUROBI on Travis, I think I am going to disable compilation of ocra software in the codyco-superbuil Travis, what do you think?

jeljaik, 2017-05-09:
My bad! ☝️There's only one client needing Gurobi, it's not the entire library. I'll put the corresponding cmake flag once I get home (Colombia) and merge accordingly. Sorry for the hiccup!

traversaro, 2017-05-09:
Ok man, thanks!

rlober, 2017-05-16:
@jeljaik 

You have a chance to fix this yet, or you want me to do it?

jeljaik, 2017-05-16:
I'm on it :)

rlober, 2017-05-18:
Gracias amigo ! This will be our first commit from Colombia

jeljaik, 2017-05-18:
Alright guys, for linux at least it's fixed. I checked directly in travis and merged with the `master` branch of `ocra-wbi-plugins` but I didn't activate it yet in `codyco-superbuild` 

What's missing to solve is the Eigen thingy... I'm gonna try to play with the submodules and see how it goes. 

[image]

diegoferigo, 2017-07-04:
Any news on the ocra build status? @jeljaik @rlober

jeljaik, 2017-07-04:
We're working on separate forks of `ocra-recipes` and `ocra-wbi-plugins` to completely remove `Eigen-LGSM`.

traversaro, 2018-04-18:
Closing due to https://github.com/robotology/codyco-superbuild/issues/198 .

closed 2018-04-18 by traversaro