robotology-legacy/codyco-superbuild #176: Problem in codyco-superbuild in macOS

traversaro, 2017-06-08:
Daniele Pucci (@DanielePucci) , [8 Jun 2017, 13:40]:
~~~
Error while obtaining sizes from MEX S-function 'WBToolbox' in 'testFTSensorsLegs/DoFs converter'.
Caused by:
Invalid MEX-file '/Users/DanielePucci/src/codyco-superbuild/build/install/mex/WBToolbox.mexmaci64': dlopen(/Users/DanielePucci/src/codyco-superbuild/build/install/mex/WBToolbox.mexmaci64, 6): Library not loaded: liburdfdom_sensor.0.4.dylib
  Referenced from: /Users/DanielePucci/src/codyco-superbuild/build/install/lib/libyarpwholebodyinterface.0.3.5.dylib
  Reason: image not found
Danieles-MacBook-Pro:build DanielePucci$ otool -L /Users/DanielePucci/src/codyco-superbuild/build/install/lib/libyarpwholebodyinterface.0.3.5.dylib
/Users/DanielePucci/src/codyco-superbuild/build/install/lib/libyarpwholebodyinterface.0.3.5.dylib:
 @rpath/libyarpwholebodyinterface.0.3.5.dylib (compatibility version 0.3.5, current version 0.3.5)
 @rpath/libwhole-body-interface.0.2.dylib (compatibility version 0.2.0, current version 0.2.6)
 @rpath/libidyntree-modelio-symoro.dylib (compatibility version 0.0.0, current version 0.0.0)
 @rpath/libidyntree-modelio-ikin.dylib (compatibility version 0.0.0, current version 0.0.0)
 @rpath/libidyntree-high-level.dylib (compatibility version 0.0.0, current version 0.0.0)
 @rpath/libidyntree-icub.dylib (compatibility version 0.0.0, current version 0.0.0)
 @rpath/libidyntree-estimation-kdl.dylib (compatibility version 0.0.0, current version 0.0.0)
 @rpath/libidyntree-icub-kdl.0.4.0.dylib (compatibility version 0.4.0, current version 0.4.0)
 @rpath/libidyntree-high-level-kdl.dylib (compatibility version 0.0.0, current version 0.0.0)
 @rpath/libidyntree-visualization.dylib (compatibility version 0.0.0, current version 0.0.0)
 @rpath/libexpression-parser.dylib (compatibility version 0.0.0, current version 0.0.0)
 @rpath/libidyntree-estimation.dylib (compatibility version 0.0.0, current version 0.0.0)
 @rpath/libidyntree-yarp.dylib (compatibility version 0.0.0, current version 0.0.0)
 @rpath/libidyntree-yarp-kdl.0.4.0.dylib (compatibility version 0.4.0, current version 0.4.0)
 @rpath/libidyntree-regressors.dylib (compatibility version 0.0.0, current version 0.0.0)
 @rpath/libYARP_OS.1.dylib (compatibility version 1.0.0, current version 2.3.69)
 /usr/local/opt/ace/lib/libACE.dylib (compatibility version 0.0.0, current version 0.0.0)
 /usr/lib/libedit.3.dylib (compatibility version 2.0.0, current version 3.0.0)
 /usr/lib/libncurses.5.4.dylib (compatibility version 5.4.0, current version 5.4.0)
 @rpath/libYARP_sig.1.dylib (compatibility version 1.0.0, current version 2.3.69)
 @rpath/libYARP_math.1.dylib (compatibility version 1.0.0, current version 2.3.69)
 @rpath/libYARP_dev.1.dylib (compatibility version 1.0.0, current version 2.3.69)
 @rpath/libYARP_name.1.dylib (compatibility version 1.0.0, current version 2.3.69)
 @rpath/libYARP_init.1.dylib (compatibility version 1.0.0, current version 2.3.69)
 /usr/local/opt/gsl/lib/libgsl.0.dylib (compatibility version 18.0.0, current version 18.0.0)
 /usr/local/opt/gsl/lib/libgslcblas.0.dylib (compatibility version 1.0.0, current version 1.0.0)
 @rpath/libidyntree-modelio-urdf-kdl.dylib (compatibility version 0.0.0, current version 0.0.0)
 @rpath/libidyntree-modelio-urdf.dylib (compatibility version 0.0.0, current version 0.0.0)
 @rpath/libidyntree-sensors.dylib (compatibility version 0.0.0, current version 0.0.0)
 liburdfdom_sensor.0.4.dylib (compatibility version 0.4.0, current version 0.0.0)
 liburdfdom_model_state.0.4.dylib (compatibility version 0.4.0, current version 0.0.0)
 liburdfdom_model.0.4.dylib (compatibility version 0.4.0, current version 0.0.0)
 liburdfdom_world.0.4.dylib (compatibility version 0.4.0, current version 0.0.0)
 libconsole_bridge.0.3.dylib (compatibility version 0.3.0, current version 0.0.0)
 @rpath/libidyntree-kdl.dylib (compatibility version 0.0.0, current version 0.0.0)
 @rpath/liborocos-kdl.1.2.dylib (compatibility version 1.2.0, current version 1.2.3)
 /usr/local/opt/tinyxml/lib/libtinyxml.dylib (compatibility version 0.0.0, current version 0.0.0)
 @rpath/libidyntree-model.dylib (compatibility version 0.0.0, current version 0.0.0)
 @rpath/libidyntree-core.dylib (compatibility version 0.0.0, current version 0.0.0)
 /usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 307.5.0)
 /usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1238.50.2)
Danieles-MacBook-Pro:build DanielePucci$
~~~

traversaro, 2017-06-08:
I guess is something strange related to https://github.com/robotology/codyco-superbuild/pull/174 / https://github.com/robotology/codyco-superbuild/pull/175 . 

To workaround this problem, try to compile the superbuild with the CMake `CODYCO_USES_KDL` option set to `OFF`. This would avoid to link `urdfdom` at all.

traversaro, 2017-06-08:
cc @francesco-romano @diegoferigo

diegoferigo, 2017-06-08:
Can [this error](https://travis-ci.org/robotology/codyco-superbuild/jobs/240755524#L7270) be related? (line 7270 if it doesn't go there automatically)

traversaro, 2017-06-08:
@diegoferigo that error is the confirmation of the everlasting rule : "Whenever you commit directly on `master` due to a demo, you always break everything else" : ) . 
Fixed by  https://github.com/robotology/yarp-wholebodyinterface/commit/d635f39dd537d0c32a6f959fc13a5f08967c9287 .

diegoferigo, 2017-06-08:
Kind of complementary to the 5-minutes-long black hole that mysteriously, before the demo, darkens portion of code and breaks everything ^^

closed 2017-07-04 by diegoferigo