robotology-legacy/codyco-superbuild #178: Codyco superbuild cannot compile WBIToolboxControllers (YARP missing)

gabrielenava, 2017-06-12:
I am trying to compile codyco-superbuild with YARP inside, and with the option 

`CODYCO_USES_WBI_TOOLBOX_CONTROLLERS = ON`. 

If YARP has not been installed already (e.g. it is the first time the project is built), then cmake outputs the following error:

```
[ 20%] Performing configure step for 'WBIToolboxControllers'
Not searching for unused variables given on the command line.
loading initial cache file /home/gnava/Software/codyco-superbuild/build/main/WBIToolboxControllers/CMakeFiles/YCMTmp/WBIToolboxControllers-cache.cmake
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:10 (find_package):
  By not providing "FindYARP.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "YARP", but
  CMake did not find one.

  Could not find a package configuration file provided by "YARP" with any of
  the following names:

    YARPConfig.cmake
    yarp-config.cmake

  Add the installation prefix of "YARP" to CMAKE_PREFIX_PATH or set
  "YARP_DIR" to a directory containing one of the above files.  If "YARP"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/gnava/Software/codyco-superbuild/build/main/WBIToolboxControllers/CMakeFiles/CMakeOutput.log".
CMakeFiles/WBIToolboxControllers.dir/build.make:101: recipe for target 'main/WBIToolboxControllers/CMakeFiles/YCMStamp/WBIToolboxControllers-configure' failed
make[2]: *** [main/WBIToolboxControllers/CMakeFiles/YCMStamp/WBIToolboxControllers-configure] Error 1
CMakeFiles/Makefile2:237: recipe for target 'CMakeFiles/WBIToolboxControllers.dir/all' failed
make[1]: *** [CMakeFiles/WBIToolboxControllers.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

```

An easy trick to make it compile again is to disable the option  `CODYCO_USES_WBI_TOOLBOX_CONTROLLERS`, then compile the superbuild, then enable the option again. This time YARP is found and the compilation is completed successfully.

traversaro, 2017-06-12:
There are two ways of fixing this. 
We can add the dependency on YARP in this file : https://github.com/robotology/codyco-superbuild/blob/master/cmake/BuildWBIToolboxControllers.cmake (see for example in https://github.com/robotology/codyco-superbuild/blob/master/cmake/BuildICUB.cmake). 
Otherwise we can remove the dependency of YARP from its CMake, as it seems that is not really necessary: https://github.com/robotology-playground/WBI-Toolbox-controllers/blob/master/CMakeLists.txt#L10 .

gabrielenava, 2017-06-12:
I chose the second option: I removed the yarp dependencies  (see https://github.com/robotology-playground/WBI-Toolbox-controllers/commit/598d0db1bdceac3875bfb2c5d17b77aad3082496), then I compiled codyco again without errors. I also tested the balancing simulation in Gazebo and it was working as usual, so I think we can close this issue.

closed 2017-06-12 by gabrielenava