robotology-legacy/codyco-superbuild #181: Mex can not compile c file in Mac os 10.12

binggnid, 2017-06-19:
I compile a simple c file using mex in matlab 2016b. I got error message:
`Building with 'Xcode with Clang'.
/usr/bin/xcrun -sdk macosx10.12 clang -c -DMX_COMPAT_32   -DMATLAB_MEX_FILE -I"/Applications/MATLAB_R2016b.app/extern/include" -I"/Applications/MATLAB_R2016b.app/simulink/include" -fno-common -arch x86_64 -mmacosx-version-min=10.12 -fexceptions -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -O2 -DNDEBUG /Users/hongyanlb/Documents/elegant/atzip/hello.c -o /var/folders/t1/yny497zj5qg325fl4nghh99h0000gn/T/mex_32330516501571_551/hello.o
/usr/bin/xcrun -sdk macosx10.12 clang -c -DMX_COMPAT_32   -DMATLAB_MEX_FILE -I"/Applications/MATLAB_R2016b.app/extern/include" -I"/Applications/MATLAB_R2016b.app/simulink/include" -fno-common -arch x86_64 -mmacosx-version-min=10.12 -fexceptions -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -O2 -DNDEBUG /Applications/MATLAB_R2016b.app/extern/version/c_mexapi_version.c -o /var/folders/t1/yny497zj5qg325fl4nghh99h0000gn/T/mex_32330516501571_551/c_mexapi_version.o
/usr/bin/xcrun -sdk macosx10.12 clang -Wl,-twolevel_namespace -undefined error -arch x86_64 -mmacosx-version-min=10.12 -Wl,-syslibroot,/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -bundle  -Wl,-exported_symbols_list,"/Applications/MATLAB_R2016b.app/extern/lib/maci64/mexFunction.map" /var/folders/t1/yny497zj5qg325fl4nghh99h0000gn/T/mex_32330516501571_551/hello.o /var/folders/t1/yny497zj5qg325fl4nghh99h0000gn/T/mex_32330516501571_551/c_mexapi_version.o  -O -Wl,-exported_symbols_list,"/Applications/MATLAB_R2016b.app/extern/lib/maci64/mexFunction.map"  -L"/Applications/MATLAB_R2016b.app/bin/maci64" -lmx -lmex -lmat -lc++ -o hello.mexmaci64
Error using mex
Undefined symbols for architecture x86_64:
  "_mexFunction", referenced from:
     -exported_symbol[s_list] command line option
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
 
`
Here is mexopts.sh
`        maci64)
#----------------------------------------------------------------------------
            # StorageVersion: 1.0
            # CkeyName: Clang
            # CkeyManufacturer: Apple
            # CkeyLanguage: C
            # CkeyVersion:
            # CkeyLinkerName:
            # CkeyLinkerVersion:
            CC='xcrun  -sdk macosx10.12  clang'
## workaround clang defect temporarily use line below           SDKROOT='/Developer/SDKs/MacOSX10.6.sdk'
# compute SDK root on the fly
# target 10.12
            MW_SDKROOT_TMP="find `xcode-select -print-path` -name MacOSX10.12.sdk"
#			MW_SDKROOT=`$MW_SDKROOT_TMP`
		MW_SDKROOT='/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk'
            MACOSX_DEPLOYMENT_TARGET='10.12'
            ARCHS='x86_64'
            CFLAGS="-fno-common -arch $ARCHS -isysroot $MW_SDKROOT -mmacosx-version-min=$MACOSX_DEPLOYMENT_TARGET"
            CFLAGS="$CFLAGS  -fexceptions"
            CFLAGS="$CFLAGS -Dchar16_t=uint16_t"
            CLIBS="$MLIBS"
            COPTIMFLAGS='-O2 -DNDEBUG'
            CDEBUGFLAGS='-g'
#
            CLIBS="$CLIBS -lc++"
            # C++keyName: Clang++
            # C++keyManufacturer: Apple
            # C++keyLanguage: C++
            # C++keyVersion:
            # C++keyLinkerName:
            # C++keyLinkerVersion:
            CXX='xcrun  -sdk macosx10.12  clang++'
            CXXFLAGS="-fno-common -fexceptions -arch $ARCHS -isysroot $MW_SDKROOT -mmacosx-version-min=$MACOSX_DEPLOYMENT_TARGET -std=c++11 -stdlib=libc++ -Dchar16_t"

	    CXXLIBS="-lc++"
            CXXLIBS="$CXXLIBS $MLIBS -lc++"
            CXXOPTIMFLAGS='-O2 -DNDEBUG'
            CXXDEBUGFLAGS='-g'
#
            # FortrankeyName: Intel Fortran
            # FortrankeyManufacturer: Intel
            # FortrankeyLanguage: Fortran
            # FortrankeyVersion: 
            # FortrankeyLinkerName: 
            # FortrankeyLinkerVersion:
            FC='ifort'
            FFLAGS='-fexceptions -fpp -mp1 -fp-model source -assume bscc -D__LP64__'
            FC_LIBDIR="$IFORT_COMPILER14/compiler/lib"
            FLIBS="$MLIBS -L$FC_LIBDIR -lifcore -limf -lintlc -lirc -lsvml"
            FOPTIMFLAGS='-O2'
            FDEBUGFLAGS='-g'
#
            LD="$CC"
            LDEXTENSION='.mexmaci64'
            LDFLAGS="-arch $ARCHS -Wl,-syslibroot,$MW_SDKROOT -mmacosx-version-min=$MACOSX_DEPLOYMENT_TARGET"
            LDFLAGS="$LDFLAGS -bundle -Wl,-exported_symbols_list,$TMW_ROOT/extern/lib/$Arch/$MAPFILE"
            LDOPTIMFLAGS='-O'
            LDDEBUGFLAGS='-g'
#
            POSTLINK_CMDS='mex_file_no_quotes="${mex_file//\"/}"'
            POSTLINK_CMDS+='; xcrun install_name_tool -change libifcore.dylib @rpath/libifcore.dylib ${mex_file_no_quotes}'
            POSTLINK_CMDS+='; xcrun install_name_tool -change libimf.dylib @rpath/libimf.dylib ${mex_file_no_quotes}'
            POSTLINK_CMDS+='; xcrun install_name_tool -change libintlc.dylib @rpath/libintlc.dylib ${mex_file_no_quotes}'
            POSTLINK_CMDS+='; xcrun install_name_tool -change libirc.dylib @rpath/libirc.dylib ${mex_file_no_quotes}'
            POSTLINK_CMDS+='; xcrun install_name_tool -change libsvml.dylib @rpath/libsvml.dylib ${mex_file_no_quotes}'
#----------------------------------------------------------------------------
            ;;
    esac
`

Any help appreciated.
Best,
Bing

traversaro, 2017-06-19:
Hi @binggnid , did you got this error by running the `codyco-superbuild` with the `CODYCO_USES_MATLAB` option set to `ON`?

binggnid, 2017-06-19:
Thank you!
I am a  noob. I don't know how to do it.

traversaro, 2017-06-19:
Did you follow the instructions in https://github.com/robotology/codyco-superbuild#macos ?

traversaro, 2017-07-03:
Hi @binggnid , feel free to reopen this issue if you still have problems.

closed 2017-07-03 by traversaro