robotology-legacy/codyco-superbuild #182: codyco superbuild with yarp-matlab-bindings  does not compile on linux

gabrielenava, 2017-07-03:
Time for my monthly report on codyco superbuild. I tried to compile codyco superbuild in `master` branch, with option `CODYCO_USES_MATLAB = ON`, and cmake retrieved the following error while compiling `yarp-matlab-bindings`:

```
[ 97%] Performing build step for 'yarp-matlab-bindings'
[ 98%] Performing install step for 'yarp-matlab-bindings'
make[3]: *** No rule to make target 'install'.  Stop.
CMakeFiles/yarp-matlab-bindings.dir/build.make:72: recipe for target 'libraries/yarp-matlab-bindings/CMakeFiles/YCMStamp/yarp-matlab-bindings-install' failed
make[2]: *** [libraries/yarp-matlab-bindings/CMakeFiles/YCMStamp/yarp-matlab-bindings-install] Error 2
CMakeFiles/Makefile2:2211: recipe for target 'CMakeFiles/yarp-matlab-bindings.dir/all' failed
make[1]: *** [CMakeFiles/yarp-matlab-bindings.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

```

I found also a possible workaround: I went in the folder `$CODYCO_ROOT/build/libraries/yarp-matlab-bindings`,

then I typed `ccmake ./` and, given the name of the repository (yarp-matlab-bindings), I enabled the option `YARP_USES_MATLAB` that was set to `OFF` by default. This fixed the problem. Waiting for the confirmation that this procedure is correct.

traversaro, 2017-07-03:
Thanks for the report, should be fixed by https://github.com/robotology/codyco-superbuild/issues/183 .

diegoferigo, 2017-07-06:
We should put an error message when we reach this state. I got stuck exactly here today, and the reason is that I build and install `yarp-matlab-bindings` externally to the `codyco-superbuild` and I forgot to enable `YARP_USES_MATLAB` (that requires a working `matlab` binary).

traversaro, 2017-07-06:
Mhhh I am a bit confused. This is an error that the `codyco-superbuild` used to print when `CODYCO_USES_MATLAB` was set to ON. If `yarp-matlab-bindings` is installed "externally" and  `CODYCO_USES_MATLAB` is set to OFF, you do not see this problem. 
On the other hand, if `yarp-matlab-bindings` is installed externally and `CODYCO_USES_MATLAB` is set to ON, then the codyco-superbuild will anyhow download and install its own version of `yarp-matlab-bindings`, as `yarp-matlab-bindings` do not install the CMake  configuration file to be found by `find_package(yarp-matlab-bindings)`.

diegoferigo, 2017-07-06:
Yes that is exactly the problem. Before discovering this issue, I had to comment out the `find_or_build_package(yarp-matlab-bindings)`. As far as I understood, the `CODYCO_USES_MATLAB` variable is used also for enabling other features beyond `yarp-matlab-bindings`, and despite I have them installed externally, I had to keep `CODYCO_USES_MATLAB=ON`. This plus the missing `YARP_USES_MATLAB` triggered the problem.

gabrielenava, 2018-01-24:
Now `yarp-matlab-bindings` is compiled correctly, should we close this issue?

traversaro, 2018-01-24:
Yes. Feel free to open again if necessary.

closed 2018-01-24 by gabrielenava