robotology-legacy/codyco-superbuild #184: CMake's CODYCO_USES_WB_TOOLBOX variable

diegoferigo, 2017-07-07:
This variable is not set anywhere, but used in [BuildWBToolbox.cmake](https://github.com/robotology/codyco-superbuild/blob/master/cmake/BuildWBToolbox.cmake#L14). Should we add an option for this?

cc @traversaro @francesco-romano @gabrielenava

traversaro, 2017-07-07:
I think it is just a leftover of the ancient past. I am ok with removing it.

traversaro, 2017-07-07:
Also `CODYCO_TRAVIS_CI` is unused in WB-Toolbox https://github.com/robotology/WB-Toolbox/search?utf8=%E2%9C%93&q=CODYCO_TRAVIS_CI&type= , so we can safely remove it as well.

traversaro, 2018-04-19:
The `robotology-superbuild'`s https://github.com/robotology/robotology-superbuild/blob/master/cmake/BuildWBToolbox.cmake is not affected by this problem.

closed 2018-04-19 by traversaro