robotology-legacy/codyco-superbuild #186: Start using the icub-models repository to provide iCub models to superbuild software

traversaro, 2017-08-24:
Until now, all the software used by the codyco-superbuild used the models installed by the yarpWholeBodyInterface . Recently we introduced the [`icub-models`](https://github.com/robotology-playground/icub-models) repository, contained only the iCub models automatically generated by [`icub-model-generator`](https://github.com/robotology-playground/icub-model-generator). 
The two repositories install their model in the same location respecting the YARP guidelines for installing robot-specific files: http://www.yarp.it/yarp_data_dirs.html . 
However, several manual modifications accumulated in the models committed in yarpWholeBodyInterface models. Furthermore there are some models that are present in yarpWholeBodyInterface that are not present in icub-models .
I think we should proceed in this way: 
* Test the real robot models present both in yarpWholeBodyInterface and icub-models to make sure that demos work fine with icub-models models. Once this has been assessed, add icub-models to the codyco-superbuild and remove from yarpWholeBodyInterface the models already contained in icub-models . 
* Understand how to handle the icubGazeboSim model, that is a model roughly matching the icub model contained in https://github.com/robotology/icub-gazebo/tree/master/icub . 
* Understand how to handle the remaining models present in the repo. 

cc @DanielePucci @fiorisi @diegoferigo @gabrielenava @nunoguedelha

gabrielenava, 2017-08-28:
Related PR: https://github.com/robotology/codyco-superbuild/pull/187

traversaro, 2018-04-18:
Moved to https://github.com/robotology/robotology-superbuild/issues/14 .

closed 2018-04-18 by traversaro