robotology-legacy/codyco-superbuild #196: CMake error: codyco-modules_INCLUDEDIR does not exist

rlober, 2018-02-08:
Hey guys,

Trying to build superbuild from scratch and I get the following cmake error: 

```
 CMake Error at /Users/Ryan/Code/codyco-superbuild/build/install/lib/cmake/codyco-modules/codyco-modulesConfig.cmake:13 (message):
   File or directory referenced by variable codyco-modules_INCLUDEDIR does not
   exist !
 Call Stack (most recent call first):
   /Users/Ryan/Code/codyco-superbuild/build/install/lib/cmake/codyco-modules/codyco-modulesConfig.cmake:19 (set_and_check)
   /Users/Ryan/Code/codyco-superbuild/build/install/share/YCM/modules/FindOrBuildPackage.cmake:180 (find_package)
   CMakeLists.txt:112 (find_or_build_package)
```

I am on 52bcb48758c8c92074fa8971e615cf6897de1984

traversaro, 2018-02-08:
Hi @rlober ! 
Can you copy&paste the content of the file `/Users/Ryan/Code/codyco-superbuild/build/install/lib/cmake/codyco-modules/codyco-modulesConfig.cmake` ? 

Depending on your needs, you also can check out the new robotology-superbuild : 
https://github.com/robotology/robotology-superbuild#which-are-the-differences-between-the-robotology-superbuild-and-the-codyco-superbuild- . It should already provide (almost) everything that we currently provide in the codyco-superbuild .

rlober, 2018-02-08:
Hey @traversaro 

Sorry about the issue. It turns out that I had an old `.bash_profile` being sourced on my mac which was messing up the paths. The cmake runs fine now. 

I will checkout the new `robotology-superbuild`. Thanks for the tip!

closed 2018-02-08 by rlober