robotology-legacy/codyco-superbuild #198: [Announcement] codyco-superbuild is deprecated in favor of the robotology-superbuild

traversaro, 2018-04-18:
Hi @robotology/codyco-developers , 
   the good old codyco-superbuild is currently being deprecated in favor of the new [`robotology-superbuild`](https://github.com/robotology/robotology-superbuild), in particular with the option [`ROBOTOLOGY_ENABLE_DYNAMICS`](https://github.com/robotology/robotology-superbuild#profile-cmake-options).

To understand more about the differences between the `codyco-superbuild` and the new `robotology-superbuild`, please check the [`robotology-superbuild`'s FAQs](https://github.com/robotology/robotology-superbuild#which-are-the-differences-between-the-robotology-superbuild-and-the-codyco-superbuild-).

Most of the workflows used on the `codyco-superbuild` should be available in the `robotology-superbuild`, if this is not the case, please [open an issue]() in the  `robotology-superbuild` repo. 

The deprecation timeline is the following: codyco-superbuild will not be updated to work on Ubuntu 18.04 . On **June 1st 2018**, the repository will be moved to the [`robotology-legacy`](https://github.com/robotology-legacy) and archived.

traversaro, 2018-05-28:
@serena-ivaldi @pi-q I noticed in https://github.com/robotology/icub-support/issues/658 that you are still using the `codyco-superbuild` . 

Are you aware of this message or for any reason you missed it? 
Do you need any help in the migration?

pi-q, 2018-05-31:
Hi @traversaro,

Thanks for caring! ^_^
We are working on updating everything, I will let you know in case we need help on that.

traversaro, 2018-06-04:
Due to some migration effort that is still ongoin, the archival of codyco-superbuild has been moved do **July 15th 2018**.

serena-ivaldi, 2018-06-06:
So we need to re-install everything using the robotlogy-superbuild? No more individual yarp, icub-main etc?

serena-ivaldi, 2018-06-06:
And what about the pc104? Do we need to install it using the superbuild there as well? Or we just install yarp and icub-main the old way (with the /build-pc104 folder)?

nunoguedelha, 2018-06-06:
Hi @serena-ivaldi , ...
- you're still allowed to have **Yarp** and **icub-main**
- you have to install the `robotology-superbuild` in the head too.

traversaro, 2018-06-06:
Hi @serena-ivaldi , 
 for normal users as mentioned by @nunoguedelha  I would suggest to use `robotology-superbuild` as document without an external YARP and ICUB installation. However, having YARP and ICUB installed externally is still perfectly supported. 

For the pc104, we are still testing locally on some robots the new workflow, so for the time being I suggest to follow the instructions in http://wiki.icub.org/wiki/Compilation_on_the_pc104 and only use the `robotology-superbuild` for the repository that are not mentioned in the guide, as you typically did for the `codyco-superbuild` .

traversaro, 2020-02-01:
> The deprecation timeline is the following: codyco-superbuild will not be updated to work on Ubuntu 18.04 . On June 1st 2018, the repository will be moved to the robotology-legacy and archived.

With just some delay, this is going to happen today. : )

closed 2020-02-01 by traversaro